adafruit/Adafruit_nRF52_Arduino #339: Compile Issue

shtarbanov, 2019-09-19:
I receive an error whenever I try compiling the example "StandardFirmataBLE" on Arduino 1.8.9 and 1.8.10.  I have tried it on both Linux Mint and Windows and for versions 0.12.0 and 0.13.0 without success for nrf52832 Feather board.

Error compiling for board Adafruit Bluefruit Feather nRF52832

hathach, 2019-09-19:
can you tell us what is your firmata library version, make sure you use this example
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/StandardFirmataBLE/StandardFirmataBLE.ino

and post your compile log error within quote here.

shtarbanov, 2019-09-19:
This is the error log on Windows, after running the latest code from Github:

C:\Users\...\Documents\Arduino\libraries\Servo\src\nrf52\Servo.cpp: In member function 'void Servo::writeMicroseconds(int)':

C:\Users\...\Documents\Arduino\libraries\Servo\src\nrf52\Servo.cpp:92:12: error: 'g_APinDescription' was not declared in this scope

  instance=(g_APinDescription[pin].ulPWMChannel & 0xF0)/16;

            ^~~~~~~~~~~~~~~~~

C:\Users\...\Documents\Arduino\libraries\Servo\src\nrf52\Servo.cpp: In member function 'int Servo::readMicroseconds()':

C:\Users\...\Documents\Arduino\libraries\Servo\src\nrf52\Servo.cpp:123:12: error: 'g_APinDescription' was not declared in this scope

  instance=(g_APinDescription[pin].ulPWMChannel & 0xF0)/16;

            ^~~~~~~~~~~~~~~~~

Multiple libraries were found for "Servo.h"
 Used: C:\Users\...\Documents\Arduino\libraries\Servo
 Not used: C:\Users\...\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.13.0\libraries\Servo
 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
exit status 1
Error compiling for board Adafruit Bluefruit Feather nRF52832.

hathach, 2019-09-19:
remove that Servo library you installed `Documents\Arduino\libraries\Servo`, it is not compatible with this BSP. This BSP comes with its own Servo library

closed 2019-09-19 by hathach

tombertrand, 2020-09-25:
I had the same error as @shtarbanov, upon deleting the base `\Servo` library, I got this new one using https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/339#issuecomment-532956767.

```
Arduino: 1.8.14 Hourly Build 2020/09/23 10:35 (Mac OS X), Board: "Adafruit Circuit Playground Bluefruit, 0.3.2 SoftDevice s140 6.1.1, Level 0 (Release)"

In file included from /var/folders/0k/2b0fbg_d0d7_nzkcmr1dw8140000gp/T/arduino_modified_sketch_992520/StandardFirmataBLE.ino:25:
/Applications/Arduino.app/Contents/Java/libraries/Firmata/Firmata.h:131:17: error: friend declaration of 'void encodeByteStream(size_t, uint8_t*, size_t) const' specifies default arguments and isn't a definition [-fpermissive]
  131 |     friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes = 0) const;
      |                 ^~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Adafruit Circuit Playground Bluefruit.
```