adafruit/Adafruit_nRF52_Arduino #343: [Bug]BLEHidAdafruit::keyboardReport has cost long time more than 100ms

shaoyuan1943, 2019-09-28:
Hi~
I'm using a board to build bluetooth keyboard. 

When I press and release a key frequently, sometimes I find ```BLEHidAdafruit::keyboardReport``` this method takes a long time more than 100ms.

my code like this:

```
void loop() {
  for (int row_num = 0; row_num < 5; ++row_num) {
    for (int col_num = 0; col_num < 10; ++col_num) {
      // check io signal
     }
   }
	
  if (key_changed) {
    auto step1 = millis();
    bool reported = BLEHidAdafruit.keyboardReport(perph_conn_handle, mods, report);
    auto step2 = millis();
  }
	
  waitForEvent();
}
```

It seems that this problem is a bit difficult to explain, I can't finf any documentation about this problem, has anyone encountered this problem?

Thank all~

shaoyuan1943, 2019-09-30:
Sorry，is my fault. Please close this issue.

closed 2019-09-30 by shaoyuan1943

dzervas, 2020-06-25:
I'm facing the same issue - ~50ms to send a `keyboardReport`. What was your problem? How did you solve it? @shaoyuan1943

shaoyuan1943, 2020-07-01:
@dzervas I'm already gave up...