adafruit/Adafruit_nRF52_Arduino #344: [Bug] BlueFruit52 library -- uninitialized variable use

henrygab, 2019-09-30:
@hathach -- my review strongly suggests that this is a bug, caused by the use of uninitialized memory.  I'd appreciate your review/confirmation.

#### Describe the bug
The BlueFruit52 library included with release 0.14.0 does not properly initialize a stack-allocated variable.  Therefore, it's undefined behavior when the code executes (i.e., depends on compiler, optimizations, … etc.).

#### Set up (please complete the following information)
  - BSP: AdaFruit Bluefruit Feather nRF52840 Express, v0.14.0
  - OS Version: Windows 10
  - Serial debug log: not applicable … see details below

####To Reproduce
Compile with all warnings enabled in Arduino.  For example, this may cause the following compiler flags to be set:
`-Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-pointer-arith`

Alternatively, manually review the code as follows:
1. Note the following code:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/fc6372836b6f2c31fc96d8ad9bc6670f8af0cc47/libraries/Bluefruit52Lib/src/BLEGatt.cpp#L214-L216
    a. At line 214, structure `ble_gattc_handle_value_t` is allocated on the stack, but not initialized
    b. At line 216, that stack-allocated structure is the second parameter to the function call
2. Looking at that function's implementation:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/fc6372836b6f2c31fc96d8ad9bc6670f8af0cc47/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/ble_gattc.h#L688-L692
    a. At line 688, note the second parameter is named  `p_iter` within this function
    b. At line 692, note that the value of `p_iter->value` is _**read**_ .
    c. At line 692, the value of `p_next` is set conditionally based on the read value from `p_iter->value`, which as noted previously, has not been initialized.

The stack variable here may contain prior stack location values.   However, it appears that the program behavior is not defined by the source code, due to this potential use of uninitialized memory.

####Expected behavior
Variables must be initialized prior to being relied upon.
Here, I am not familiar enough with the code to be 100% sure of the fix.  However, it may be as simple as doing one of the following:
1. At BLEGatt.cpp line 214, chainging the following line ...
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/fc6372836b6f2c31fc96d8ad9bc6670f8af0cc47/libraries/Bluefruit52Lib/src/BLEGatt.cpp#L214-L216
… to explicitly initialize the structure's `value` field to null.

Wookie1, 2019-10-01:
What kind of issue did you have that pointed you to this?

henrygab, 2019-10-01:
I did not root cause any existing issue to this as the cause.  I build with warnings enabled, because I want any code I provide others to compile cleanly.  v0.14.0 caused some build warnings to appear, so I tracked them down and fixed them.  Pull request incoming...

PR #347 (2019-10-09): Compile without warnings
This PR allows v0.14.0 to compile cleanly (no warnings), allowing end-users to focus on issues within their own code.
* Fixes #344 - Use of uninitialized variable in Bluefruit library
* Fixes #345 - InternalFileSystem compiler warnings for comparing signed vs. unsigned
* Fixes #346 - Additional Bluefruit library compiler warnings

Note: I am not sure how to fully test these.  I ran the bleuart example, and verified UART capability. 

Of particular note is the changes to the InternalFileSystem library's flash_cache.c, function flash_cache_read().  Significant comments w [... cut]
closed 2019-10-09 by hathach