adafruit/Adafruit_nRF52_Arduino #352: [Bug] Why does PDM require manual IRQ triggering?

ladyada, 2019-10-12:
Ideally for the PDM library (adapted from Arduino's) we should have the IRQ auto-called when there's more data, but to make the example work, i have to manually call it.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/PDM/examples/PDMSerialPlotter/PDMSerialPlotter.ino#L44

 Why is that - what is not set up to make the IRQ enabled? That is your mystery :D

henrygab, 2019-10-14:
@ladyada - It appears the library code for the PDM peripheral may have some significant bugs in how it handles buffers.  Please accept the below code review of the library's configuration and use of the PDM peripheral.  I would appreciate confirmation of my findings.

**_Unfortunately, I do not have an appropriate microphone, and thus cannot test the results._**  However, if you wish, I can generate the fixes that appear to be necessary, and submit a PR for your own testing.

<details><summary>Interaction between tasks, events, buffers</summary>

<P>
According to the product specifications for nRF52840, v1.1 @ line 236:

> As soon as the STARTED event is received, the firmware can write
> the next SAMPLE.PTR value (this register is double-buffered), to
> ensure continuous operation.
> When the buffer in RAM is filled with samples, an END event is
> triggered. The firmware can start processing the data in the buffer.
> Meanwhile, the PDM module starts acquiring data into the new buffer
> pointed to by SAMPLE.PTR, and sends a new STARTED event, so that the
> firmware can update SAMPLE.PTR to the next buffer address.

Putting this in plain English, the rules for the code (library) appear to be:
1. Before triggering `NRF_PDM_TASK_START`, set the `SAMPLE.PTR` value.
   (_The peripheral saves (double-buffers) the `SAMPLE.PTR` register
   for immediate use, and generates a `START` event._)
2. When receiving a `START` event, the `SAMPLE.PTR` should be set to
   the *next* buffer that will be used, after the double-buffered value
   that was set (and is the current value) is filled by the peripheral.
   (_The peripheral will ignore the value place in that register
   until **after** it fills the current buffer._)
3. When receiving an `END` event, that means the peripheral finished
   filling a buffer (which was double-buffered, and thus unlikely to
   be the same as currently in the `SAMPLE.PTR` register).

</details>

<details><summary>Library fails to set `SAMPLE.PTR` prior to triggering `NRF_PDM_TASK_START`</summary>

<P>

`PDMClass::begin()` configures the peripheral, but then curiously has
the following commented-out lines:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/2c5d06993f2464437fb54e20b91b8c1593c5dfd6/libraries/PDM/src/PDM.cpp#L110-L117

As a result, when `NRF_PDM_TASK_START` is triggered at line 117, the
`SAMPLE.PTR` register has not been configured (it defaults to zero).
This results in undefined (but definitely non-desired) behavior:

> If the destination address set in SAMPLE.PTR is not pointing to the
> Data RAM region, an EasyDMA transfer may result in a HardFault or
> RAM corruption.
Address zero is *NOT* part of the DATA RAM region, and thus it appears
that at least undefined behavior will occur for the first set of samples
(i.e., a bug).

Correcting this portion appears as simple as uncommenting
lines 111..112 so that `SAMPLE.PTR` is set prior to triggering the `START` task.

</details>
<details><summary>Library incorrectly handles events</summary>

<P>

As noted above, the `NRF_PDM_EVENT_STARTED` event only means
that the peripheral has double-buffered the buffer pointer, and thus
that the pointer in `SAMPLE.PTR` can be updated to reflect a next
buffer to be used.

The code, however, does something different:
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/2c5d06993f2464437fb54e20b91b8c1593c5dfd6/libraries/PDM/src/PDM.cpp#L182-L194

* At line 182, it clears the event.  Good.
* At line 184, it checks to ensure there is a next buffer available for use.  Good.
* At line 186, it sets `SAMPLE.PTR` to point to the next buffer.  Good.
* At line 189, the code decides that the prior buffer is ready for reading.  **_Bad_**.  As noted in the product specifications, the `END` event is the one that indicates the buffer has been filled, not the `START` event.
* At lines 192..193, calls into the callback() to indicate data was received … except as noted, that is signified by the `END` event, not the `STARTED` event.

Thus, it appears that the code will need to be modified to at least move the functionality associated with lines 189..193 from the `STARTED` event handler to the `END` event handler.

</details>
<details><summary>PDMDoubleBuffer class never gets updated valid data length?</summary>

<P>

The `PDMDoubleBuffer` class (_stored in `_doubleBuffer` member variable_) appears to abstract away how much data is available to be read.  The bytes available to be read are stored in the class's private variable, `_length[2]` (one value for each buffer).  However, the only way to set this to a non-zero value that I could find is by calling the class's `write()` function.  That's fine for non-DMA writes, and it includes the `memcpy()` call that actually copies the data.  However, that function does not fit the PDM peripheral, which uses EasyDMA to update the memory directly, and thus has no source buffer to copy data from (the buffer already got the updated information).

Thus, it appears that there will need to be additional changes to the utility class as well, to update the class's internal state to reflect that a DMA transfer has made a certain number of bytes valid.
</details>

<details><summary>Summary of work required</summary>

<P>

1. Fix bug so that `SAMPLE.PTR` is always valid prior to triggering `START` event.
2. Move indication of buffer having been filled from `START` event to `END` event.
3. Modify the `PDMDoubleBuffer` class to add a function that can be called to
   indicate a DMA write has completed, effectively acting similar to the current
   `write()` function, without any memcpy(), and potentially `assert()`'ing that the
   bounds of the buffer are not exceeded (rather than just limiting to the bounds).

</details>

ladyada, 2019-10-14:
thanks for the 👀 i'll do another pass at the library!

henrygab, 2019-12-05:
Just wanted to give a heads-up on a library that I'm developing, specifically to simplify interfacing with EasyDMA-enabled hardware (such as many of the peripherals on the nRF series).

https://github.com/SimpleHacks/EzDmaHelper

It's just the start of a framework for now, but should greatly simplify use of EasyDMA hardware.

ladyada, 2019-12-05:
nice - i havent abandoned this, just...got busy :)

henrygab, 2020-01-06:
No problems... me too.  I'm still working on the helper library for DMA, but the LFS corruption ([current PR 397](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/397), [original attempt via 372](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/372)) on the nRF52 boards has taken much more time than I had anticipated.  :)

ladyada, 2020-01-06:
hiya btw i tried your fixes...but it would never run at all. so something mysterious has occured...i ran out of my time bucket so hopefully thach will be able to poke it into working :)

hathach, 2020-01-10:
closed by #409

closed 2020-01-10 by hathach

henrygab, 2020-01-11:
@hathach … Can you help me understand why the `NRF_PDM_EVENT_END` event doesn't need to be handled in the ISR?

I know it's a bit long, but my [earlier post](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/352#issuecomment-541486903) was split into nicely collapsed headings.  In particular, I think the hardware doesn't mean what you think it means when it sets the `NRF_PDM_EVENT_STARTED` event.

When the `NRF_PDM_EVENT_STARTED` event occurs, the buffer that's in the register has not necessarily been filled.  But, it appears the code is calling the `_onReceive()` callback anyways, even through the buffer is simultaneously in-use by the peripheral?  Thus, the callback gets called with a partially-filled buffer, which is probably not the intent?

The ISR also does not appear to handle the `NRF_PDM_EVENT_END` event, which is the event that I would have expected to have the `_onReceive()` callback called.

Am I misunderstanding some aspect of this code?

Thanks!

hathach, 2020-01-16:
@henrygab to be honest, I am new to PDM, just figure out how it works and reading the nrf PDM specs. Then I realize the current code "forgot" to re-enable IRQ after getting its sample. I haven't looked at your modification as well, and I don't think I know PDM better than you to explain this. I am kind of too happy/satisfied when seeing the original bug is fixed :).

henrygab, 2020-01-17:
@hathach  I challenge you to answer three questions, listed below.

The product specification PDF should be your source:
1. What does the PDM event `EVENTS_STARTED` mean?
2. What does the PDM event `EVENTS_END` mean?
3. In the PDM description, look for the term "double-buffered register".  Why is this double-buffering needed?

After answering those three questions, consider the existing code's flow as a timeline, tracking when events could occur, and the resulting state of:
1. PDM SAMPLE.PTR register contents
2. PDM double-buffered SAMPLE.PTR register
3. Exact number (or range of possible) bytes of data valid in each doubleBuffer register

Finally, consider when the callback routine is provided the buffer pointer as containing data....

NOTE: I am not challenging you for fun.  The PDM peripheral is just one example, but I think most EasyDMA enabled peripherals on the nRF5x series act very similarly, both on double-buffering and the meaning of events generated (although names differ).  Therefore, if you can answer the above, you will be much stronger in understanding the chip and better positioned to add additional functionality with other such peripherals.

hathach, 2020-01-30:
@henrygab Sorry for late response, I was in Lunar New Year holiday, busy cleaning up my messing house. Thank you for yet another thoughtful analysis. However, since I ste.... hmm USE code from official Arduino repo, I will leave it as it is since I have no interest in optimizing this PDM library (barely knew it), I am bug/feature-driven, I tend to write code to fix bug or add feature when needed. Not really proud of this manner, but it helps me to work on multiple projects. Therefore I will cowardly decline your challenge :cow: :cow: