adafruit/Adafruit_nRF52_Arduino #355: [Bug] homekit_lightbulb.ino no longer part of examples?

tamberg, 2019-10-14:
**Describe the bug**
In the past, Adafruit Bluefruit nRF52 Libaries > Projects > homekit_lightbulb was part of the examples, now it's gone.

**Set up (please complete the following information)**
Install the Adafruit_nRF52_Arduino board support package (BSP).

**Additional context**
It seems [this commit](https://github.com/adafruit/Adafruit_nRF52_Arduino/commit/a5bc1f71e6b9aa12d77daf97a7e61213611544ff#diff-d220cd62b6c724739430c410e7c134c8) renamed the .ino file to .ino.cpp which prevent sit from showing up while still being part of the installed source code examples.

ladyada, 2019-10-14:
ooh thats an easy fix, at least - wanna try submitting a PR? :)

tamberg, 2019-10-14:
https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/356

hathach, 2019-10-15:
Homekit sketch won't compile due to lots of API changes. The last time It worked, it only go passed the security pairing process of adding new device. I haven't got time to get it worked with normal transmission data yet. This whole homekit library is very time consuming one.

closed 2019-10-15 by hathach