adafruit/Adafruit_nRF52_Arduino #359: [Bug]Feather NRF52840 Serial USB

ericlangel, 2019-10-17:
Serial (USB) works with:
Arduino IDE
Tera Term

but it didn't work with HTerm and BlHeliSuite (sends and receive Serial Data to/from selected COM)

So, i'am not sure why some Terminals work and others not?
Where is the secret?

  - Feather 52840
  - WIN10
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)   -> Not working because HTERM doesn't receive anything

**To Reproduce**
Steps to reproduce the behavior:
Serial communication with HTERM...just a "hello" every second

**Expected behavior**
Nothing....no received Data
With Arduino IDE Monitor or Tera Term the "hello" will be displayed

For our Application the Feather NRF52840  has to work with BlheliSuite

hathach, 2019-10-17:
did your hterm terminal work with any other CDC of other mcu such as samd21/samd51 core ?

ericlangel, 2019-10-18:
cant test it. The NRF52840 is my first MCU with internal USB

i looked at the adafruit code and saw that DTR  has to be enabled for USB communication.
I will try to activate DTR (flow control) in HTerm.

But, i cant enable it in BlHeliSuite because its from another vendor. I just have to use it.

What would happen i the DTR Request in the Ada Code would return True at any Time?

bool tud_cdc_n_connected(uint8_t itf)
{
  // DTR (bit 0) active  is considered as connected

  //return tud_ready() && tu_bit_test(_cdcd_itf[itf].line_state, 0);

 return TRUE;
}

hathach, 2019-11-05:
you dont have to enable any flow control, it is standard windows cdc kernel driver, unless your terminal decide to set it manually to not connected. I don't think I could help with this issue, I don't even consider it is a valid issue for nrf52840 either.  You should contact the vendor terminal to fix this.

closed 2019-11-05 by hathach

ericlangel, 2019-11-05:
i got a special version of the Terminal from Vendor with enabled DTR Signal.
Now the Serial on NRF52840 works fine.

So my conclusion: the Serial functions needs active DTR Signal.

Without DTR Signal the Serial functions on NRF52840 doesn't work.
For me this is a Bug.