adafruit/Adafruit_nRF52_Arduino #364: [Bug] SPIM3 could conflict with Softdevice/IRQ and cause hardfault

hathach, 2019-10-23:
due to errata 198 https://infocenter.nordicsemi.com/topic/errata_nRF52840_Rev2/ERR/nRF52840/Rev2/latest/anomaly_840_198.html

Solution: allocated a fixed separated RAM block just for SPIM3 to use

related to https://github.com/adafruit/circuitpython/issues/1222#issuecomment-545200965

[image]

FleshdeckTV, 2021-09-12:
Hi, I have a problem, I am making a smart watch and using SPIM3 for the display, everything works well until there is a conflict between SPI and BLE, if it connects or disconnects during SPI data transfer it will cause a critical error, I changed SPIM3 to SPIM2, but this did not solve the problem, as I understand the connect_callback function is a BLE interrupt, and it has the highest priority, and breaks the SPI transmission. Is it possible to fix this?

hathach, 2021-09-12:
Then your issue is not related to this particular one. If you believe it is the bug of repo, please use the bug template to file your own issue and provide a simplest sketch along with reproducible steps on one of Adafruit boards which I have access to test.

FleshdeckTV, 2021-09-13:
I found the reason for the fatal errors, this is a data parser from BLE, I changed the code and so far everything is working well, but I still have a question about how SPI works and accelerating display drawing, I found information on the nordic forum that in nrf52840 the spi buffer is  16 bit, but in fact only 8bit is used due to nrf52832.  Can this buffer be increased?  And also I would like to understand how EasyDMA works, will you add it to work with displays?  it would be great, thanks

hathach, 2021-09-13:
as mentioned above, this issue is for the errata conflict between SPIM3 and SD which is not related to your issue. If you have any questions, post yours to appropriate support forum.

dhalbert, 2021-09-13:
@hathach By the way, we are noticing a newly reported problem, which is that SPIM3 is not working when VBUS is not present. See https://github.com/adafruit/circuitpython/issues/5233 and https://devzone.nordicsemi.com/f/nordic-q-a/74754/nrf52840-application-won-t-completely-run-without-power-applied-to-vbus.