adafruit/Adafruit_nRF52_Arduino #366: [Bug] macOS 10.15 Catalina — Upload fails

rei-vilo, 2019-10-26:
**Describe the bug**
Upload works on Windows 10 1903 but fails on macOS 10.15 Catalina.

**Set up (please complete the following information)**

- macOS 10.15 
- Arduino IDE 1.8.10  
- Adafruit nRF52 0.14.5, 
- Board: Adafruit Bluefruit Feather nRF52832 with Bootloader 0.2.13 SoftDevice s132 6.1.1

**To Reproduce**
Steps to reproduce the behavior:
1. Select board Adafruit Bluefruit Feather nRF52832
2. Select example Blink
3. Click on Upload
4. See error message below

> /Users/ReiVilo/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.5/tools/adafruit-nrfutil/macos/adafruit-nrfutil --verbose dfu serial -pkg /var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_302918/Blink.ino.zip -p /dev/cu.usbserial-010477B6 -b 115200 --singlebank 
> Upgrading target on /dev/cu.usbserial-010477B6 with DFU package /private/var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_302918/Blink.ino.zip. Flow control is disabled, Single bank, Touch disabled
> Opened serial port /dev/cu.usbserial-010477B6
> Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 22876
> Sending DFU start packet
> Timed out waiting for acknowledgement from device.
> 
> Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.
> Traceback (most recent call last):
> File "nordicsemi/__main__.py", line 294, in serial
> File "nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
> File "nordicsemi/dfu/dfu.py", line 200, in _dfu_send_image
> File "nordicsemi/dfu/dfu_transport_serial.py", line 179, in send_start_dfu
> File "nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
> File "nordicsemi/dfu/dfu_transport_serial.py", line 282, in get_ack_nr
> nordicsemi.exceptions.NordicSemiException: No data received on serial port. Not able to proceed.
> 
> Possible causes:
> - Selected Bootloader version does not match the one on Bluefruit device.
> Please upgrade the Bootloader or select correct version in Tools->Bootloader.
> - Baud rate must be 115200, Flow control must be off.
> - Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
> 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

Trying to update the boot-loader following the procedure [Update the nRF52832 bootloader](https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/updating-the-bootloader) fails at well.

> /Users/ReiVilo/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.5/tools/adafruit-nrfutil/macos/adafruit-nrfutil --verbose dfu serial -pkg /Users/ReiVilo/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.5/bootloader/feather_nrf52832/feather_nrf52832_bootloader-0.2.13_s132_6.1.1.zip -p /dev/cu.usbserial-010477B6 -b 115200 --touch 1200 
> Upgrading target on /dev/cu.usbserial-010477B6 with DFU package /Users/ReiVilo/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.5/bootloader/feather_nrf52832/feather_nrf52832_bootloader-0.2.13_s132_6.1.1.zip. Flow control is disabled, Dual bank, Touch 1200
> 
> Failed to upgrade target. Error is: Serial port could not be opened on /dev/cu.usbserial-010477B6. Reason: [Errno 16] could not open port /dev/cu.usbserial-010477B6: [Errno 16] Resource busy: '/dev/cu.usbserial-010477B6'
> Traceback (most recent call last):
>   File "serial/serialposix.py", line 265, in open
> OSError: [Errno 16] Resource busy: '/dev/cu.usbserial-010477B6'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "nordicsemi/dfu/dfu_transport_serial.py", line 99, in open
>   File "serial/serialutil.py", line 240, in __init__
>   File "serial/serialposix.py", line 268, in open
> serial.serialutil.SerialException: [Errno 16] could not open port /dev/cu.usbserial-010477B6: [Errno 16] Resource busy: '/dev/cu.usbserial-010477B6'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "nordicsemi/__main__.py", line 294, in serial
>   File "nordicsemi/dfu/dfu.py", line 226, in dfu_send_images
>   File "nordicsemi/dfu/dfu.py", line 157, in _dfu_send_image
>   File "nordicsemi/dfu/dfu_transport_serial.py", line 101, in open
> nordicsemi.exceptions.NordicSemiException: Serial port could not be opened on /dev/cu.usbserial-010477B6. Reason: [Errno 16] could not open port /dev/cu.usbserial-010477B6: [Errno 16] Resource busy: '/dev/cu.usbserial-010477B6'
> 
> Possible causes:
> - Selected Bootloader version does not match the one on Bluefruit device.
>     Please upgrade the Bootloader or select correct version in Tools->Bootloader.
> - Baud rate must be 115200, Flow control must be off.
> - Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
> 

**Additional context**
Very same procedure succeeds on Windows 10 1903 with Arduino IDE 1.8.10 and Adafruit nRF52 0.14.5, targeting Adafruit Bluefruit Feather nRF52832 with Bootloader 0.2.13 SoftDevice s132 6.1.1

> C:\Users\ReiVilo\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.14.5/tools/adafruit-nrfutil/win32/adafruit-nrfutil.exe --verbose dfu serial -pkg C:\Users\ReiVilo\AppData\Local\Temp\arduino_build_871429/Blink.ino.zip -p COM7 -b 115200 --singlebank 
> Upgrading target on COM7 with DFU package C:\Users\ReiVilo\AppData\Local\Temp\arduino_build_871429\Blink.ino.zip. Flow control is disabled, Single bank, Touch disabled
> Opened serial port COM7
> Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 22876
> Sending DFU start packet
> Sending DFU init packet
> Sending firmware file
> ########################################
> #####
> Activating new firmware
> 
> Device programmed.
> DFU upgrade took 3.6966779232025146s

[image]

christophepersoz, 2019-10-26:
So it fails for nRF52832 and for nRF52840. I haven't test on nRF52832 but I got this on nRF52840. I gonna try the UF2 method and see how it goes.

rei-vilo, 2019-10-29:
It looks like the [SiLabs CP2104 driver](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) needs to be installed again, even when it was already installed before updating macOS.

Updating macOS seems to break the previously installed driver, so it needs to be reinstalled. 

On my case, the CP2104 driver was installed with macOS 10.14 _Mojave_. After I updated to Mac 10.15 _Catalina_, it wasn't working properly and the Feather nRF52832 was no longer recognised. Once I reinstalled CP2104 driver, I could upload to the Feather nRF52832 again.

closed 2019-10-29 by rei-vilo

christophepersoz, 2019-10-29:
Please note that's also the case if you are upgrading from High Sierra to Mojave (10.14), the driver is not kept valid neither active after the upgrade and need to be reinstalled.

ufanders, 2020-06-08:
Hey, I just did a fresh install of OSX Mojave, with latest security update, then installed the CP2104 driver and I am experiencing a similar issue. The firmware upload seems to start, albeit extremely slowly, then fails. I am using the Bluefruit nRf52840 Feather Express board that worked well before I refreshed my Mac.

Debug info:

> Sketch uses 103956 bytes (12%) of program storage space. Maximum is 815104 bytes.
Global variables use 12724 bytes (5%) of dynamic memory, leaving 224844 bytes for local variables. Maximum is 237568 bytes.
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem21101
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBtoUART2, /dev/tty.usbmodem21101, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.SLAB_USBtoUART, /dev/cu.SLAB_USBtoUART2, /dev/cu.usbmodem21101, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.SLAB_USBtoUART, /dev/tty.SLAB_USBt [... cut ...]
  File "nordicsemi/__main__.py", line 294, in serial
  File "nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
  File "nordicsemi/dfu/dfu.py", line 200, in _dfu_send_image
  File "nordicsemi/dfu/dfu_transport_serial.py", line 179, in send_start_dfu
  File "nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
  File "nordicsemi/dfu/dfu_transport_serial.py", line 282, in get_ack_nr
nordicsemi.exceptions.NordicSemiException: No data received on serial port. Not able to proceed.

Any ideas? I'm dead in the water!

synth, 2021-01-01:
Just a shot - have you tried changing usb cables? A while ago, I struggled to connect to my esp and then happened to have a somehow uniquely special micro-usb cable and I was able to upload to it. [Many other people reported](https://github.com/espressif/arduino-esp32/issues/1084#issuecomment-751387906) that it was the cable as well. 

I moved and lost that cable and now that I'm trying to connect again, I can't seem to be able to, despite reinstalling drivers and rebooting and many other attempts. I wish it wasn't this hard to just get connected on a Mac :/

simonjcarr, 2021-01-24:
I have the same problem, I install the drivers and the USB port appears, I flash the device, I disconnect and reconnect to the device and the USB port disappears.

The solution is re-run the Silicon Labs install. The installer sees that the software is already installed and just refreshes the driver cache and the port is there again. 

I think it must be a bug in the Silicon Labs software (driver cache).

hathach, 2021-01-25:
There is some weird issues between Silabs driver and macOS, You may need to unblock it from system extension and/or security feature last time I installed. It may be a bit different with latest macOS and latest Siliabs.
https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/arduino-board-setup