adafruit/Adafruit_nRF52_Arduino #367: BLE Central example with multiple NOTIFY characteristics

jeromeDevIN, 2019-10-30:
Hello,

I am working on two programs based on the Custom HRM example (including both Peripheral and Cental).

I would like to program a Service with two Characteristics with Notify property.

In the Adafruit example, the HRM Peripheral has one Service (hrms) and one characteristic (hrmc) wih the NOTIFY property.

In my Peripheral program, I have set two characteristics (with proprietary UUID) with NOTIFY property in one Service (proprietary UUID). 

Here is the essential part of my two programs :

### Peripheral program

I declare the Service and Characteristics UUID

```
//Characteristics
#define UUID16_CHR_ENERGY_MEASUREMENT 0x4601
#define UUID16_CHR_ENOUGH_ENERGY_WORK 0x4602
#define UUID16_CHR_STAND_BY 0x4603
```

Then I declare the BLEService and BLECharacteristic objects
```
BLEService        sls = BLEService(UUID_SVC_SMART_LAYER);  //SMART LAYER SERVICE
BLECharacteristic emc = BLECharacteristic(UUID16_CHR_ENERGY_MEASUREMENT); //ENERGY MESUREMENT
BLECharacteristic eewc = BLECharacteristic(UUID16_CHR_ENOUGH_ENERGY_WORK); //ENOUGH ENERGY TO WORK
```

The init function of the Service and characteristics in the Peripheral is as follow:

```
...
  void setupHRM(void)
  {
    sls.begin();

    emc.setProperties(CHR_PROPS_NOTIFY);
    emc.setPermission(SECMODE_OPEN, SECMODE_NO_ACCESS);
    emc.setFixedLen(2);
    emc.setCccdWriteCallback(cccd_callback);  // Optionally capture CCCD updates
    emc.begin(); 

    eewc.setProperties(CHR_PROPS_NOTIFY);
    eewc.setPermission(SECMODE_OPEN, SECMODE_NO_ACCESS);
    eewc.setFixedLen(1);
    eewc.setCccdWriteCallback(cccd_callback);  // Optionally capture CCCD updates
    eewc.begin(); 
...

  }
```

Then the CCCD callback function is : 

```

void cccd_callback(uint16_t conn_hdl, BLECharacteristic* chr, uint16_t cccd_value)
{    
    // Display the raw request packet
    Serial.print("CCCD Updated: ");
    //Serial.printBuffer(request->data, request->len);
    Serial.print(cccd_value);
    Serial.println("");

    // Check the characteristic this CCCD update is associated with in case
    // this handler is used for multiple CCCD records.
    if (chr->uuid == emc.uuid) {
        if (chr->notifyEnabled(conn_hdl)) {
            Serial.println("Energy Measurement 'Notify' enabled");
        } else {
            Serial.println("Energy Measurement 'Notify' disabled");
        }
    } 
    
    else if (chr->uuid == eewc.uuid) {
      if (chr->notifyEnabled(conn_hdl)) {
          Serial.println("Enough Energy to Work 'Notify' enabled");
      } else {
          Serial.println("Enough Energy to Work 'Notify' disabled");
      }
    }
}
```
In the loop function, I notify the two characteristics value to the Central : 

```
...
  if ( emc.notify16(energyMeasure) ){
      Serial.print("Energy Measurement updated to: "); Serial.println(energyMeasure); 
  }else{
      Serial.println("ERROR: Energy Measurement Notify not set in the CCCD or not  connected!");
  }
...
  if ( eewc.notify8(enoughEnergy) ){
      Serial.print("Enough Energy to Work updated to: "); Serial.println(enoughEnergy); 
  }else{
      Serial.println("ERROR: Enough Energy to Work Notify not set in the CCCD or not connected!");
  }
...
```

### Central program

Here is the Service Init part in the setup() function:

```
  ...  
  // Initialize SL client
  sls.begin();  
  emc.setNotifyCallback(notify_callback, true);
  emc.begin();

  sls.begin();    
  eewc.setNotifyCallback(notify_callback, true);
  eewc.begin();   
  ...
``` 

Then the connect_callback function:

```
void connect_callback(uint16_t conn_handle)
{  
  
  Serial.println("Connected");
  Serial.print("Discovering Smart-Layer Service ... ");

  // If Service sls is not found, disconnect and return
  if ( !sls.discover(conn_handle) )
  {
    Serial.println("Found NONE");
    // disconnect since we couldn't find SL service
    Bluefruit.disconnect(conn_handle);
    return;
  }

  // Once sls service is found, we continue to discover its characteristics
  Serial.println("Found it");

  Serial.print("Discovering Energy Measurement characteristic ... ");
  if ( !emc.discover() )
  {    
    Serial.println("not found !!!");
    Serial.println("Energy Measurement characteristic is mandatory but not found");
    Bluefruit.disconnect(conn_handle);
    return;
  }
  else {
    Serial.println("Found it");
    // Reaching here means we are ready to go, let's enable notification on measurement chr
    if ( emc.enableNotify() )
    {
      Serial.println("Ready to receive Smart-Layer Energy Measurement value");
    } else
    {
      Serial.println("Couldn't enable notify for Smart-Layer Energy Measurement. Increase DEBUG LEVEL for troubleshooting");
    }  
  }
  
  Serial.print("Discovering Energy Enough to Work characteristic ... ");
  if ( !eewc.discover() )
  {
    Serial.println("not found !!!");
    Serial.println("Enough Energy to Work characteristic is mandatory but not found");
    Bluefruit.disconnect(conn_handle);
    return;
  }
  else{  
    Serial.println("Found it");    
    // Reaching here means we are ready to go, let's enable notification on measurement chr
    if ( eewc.enableNotify() )
    {
      Serial.println("Ready to receive Smart-Layer Enough Energy to Work value");
    } else
    {
      Serial.println("Couldn't enable notify for Smart-Layer Enough Energy to Work. Increase DEBUG LEVEL for troubleshooting");
    }      
  }  
}
```

And then, the callback function called when the central get a notification from the both characteristics:

```
void notify_callback(BLEClientCharacteristic* chr, uint8_t* data, uint16_t len)
{

   if (emc.uuid == chr->uuid){
      Serial.print("Energy Measurement: ");
      uint16_t valueEnergy;
      memcpy(&valueEnergy, data, 2);
    
      Serial.print("Energy measurement received: ");
      Serial.println(valueEnergy);
  }
  
  if (eewc.uuid == chr->uuid){
    Serial.print("Energy Enough to Work: ");
    uint8_t valueEnoughE;
    memcpy(&valueEnoughE, data, 1);
    
    Serial.print("Energy Enough to Work received: ");
    Serial.println(valueEnoughE);
  }
}

```

I get the two notified value with the nRF Connect software one a PC or smartphone as well.

But my Central program only gets the notified value of only one characteristic. 

The Peripheral monitor displays : "ERROR: Energy Measurement Notify not set in the CCCD or not connected!" at the call of `emc.notify16(energyMeasure)`

And the Central only gets one characteristic value notified. The Central only displays : 

`"Enough Energy to Work 'Notify' enabled". `
It seems that the ch->uuid is always the same (eewc.uuid) in the notify_callback call.

Could you help me on this case, just give me the right sequence of call, the right instructions that works ?

Best regards,

Jerome

orhanyor, 2020-01-30:
i noticed you call sls.begin(); twice in your client sketch is there any reason for that?
also did you try creating individual notify functions like:
emc.setNotifyCallback(emc_notify_callback);
and 
eewc.setNotifyCallback(eewc_notify_callback);

and then down below create 2 different functions for emc_notify_callback and eewc_notify_callback. at least thats how it is implemented in the examples. its worth giving a try.

jeromeDevIN, 2020-02-11:
Hello,

Your advises worth the try. I have used two callback functions for the two characteristics:

emc.setNotifyCallback(emc_notify_callback);
and
eewc.setNotifyCallback(eewc_notify_callback);

and it works better. The central get the notification from the two characteristics.

Thanks for the help

hathach, 2020-02-11:
Thanks for confirmation

closed 2020-02-11 by hathach