adafruit/Adafruit_nRF52_Arduino #375: Serial output from CP Bluefruit blocks when not connected

jhmaloney, 2019-11-05:
**Describe the bug**

On the Circuit Playground Bluefruit Alpha, the serial output buffer appears to fill up when the board is not connected to a computer via a USB cable. This causes applications that print or log data over USB to eventually block waiting for space to become available for writing when they are not connected to a USB cable.

Is this the intended behavior? Virtually every other microcontroller I've used from AdaFruit and others will continue to transmit serial data whether the board is connected to a USB cable or not; the hardware UART doesn't care...

More detail:

The symptom of the serial output buffer being full is that Serial.write(aByte) returns 0, indicating that the byte was not written. While it is normal for the serial buffer to fill up when data is written faster than it can be transmitted, normally space eventually becomes available again, allowing the next byte to be written. That process works as expected when the CP Bluefruit is connected to a computer with a USB-serial cable and an application such as the Arduino IDE's console or a terminal program is reading data from the serial port.

However, when the CP Bluefruit is not connected to a computer --nor when no application on the computer is consuming data from the serial port -- then once the output buffer is full, output buffer space never becomes available. This is not the case for the CP Express, the BBC micro:bit, the AdaFruit ItsyBitsy M0, Gemma M0, Trinket M0, ESP32, ESP8266, or any of the many Arduino boards I own, so I'm guessing this is not the intended behavior.

**Set up (please complete the following information)**
  - Board: Circuit Playground Bluefruit - ALPHA (https://www.adafruit.com/product/4333)
  - MacOS 10.14.6 (Mojave)

jhmaloney, 2019-11-05:
> can you provide your sketch and steps to reproduce it.

I've seen the symptoms in MicroBlocks (http://microblocks.fun) but that's a big chunk of code.

I'll throw together a small Arduino sketch to demo the problem. Might take me a day or two...

	-- John

jhmaloney, 2019-11-06:
Here's the test program:

`void setup() {
	pinMode(LED_BUILTIN, OUTPUT);
	Serial.begin(115200);
}

void loop() {
	digitalWrite(LED_BUILTIN, HIGH);
	while (!Serial.write(65)); // wait until Serial.write() sends a byte
	delay(100);
	digitalWrite(LED_BUILTIN, LOW);
	delay(100);
}
`
On many boards, this function returns whether or not the board is connected to an app that is consuming data from the serial port. On the Circuit Playground Bluefruit, however, it blocks if not connected to an app that is consuming data. It also blocks if the board is running on batteries and has no USB connection at all. That was when I noticed the problem -- I'd written a program to both record and report sensor data and it hung when the serial output buffer filled up.

But the non-blocking behavior I saw on the Circuit Playground Express ItsyBitsy M0 is harder to reproduce than I thought. After spending a day trying all sorts of things I finally discovered that the behavior depends on both the toolchain and the framework version.

When compiled with PlatformIO and 4.3.190711 of the arduinosam package (https://github.com/platformio/platformio-pkg-framework-arduinosam"), the code doesn't block on the AdaFruit SAMD21 boards I have (CIrcuit Playground Express and ItsyBitsy M0), while it does block on the Circuit Playground Bluefruit. However, it blocked on all the boards when compiled with the Arduino IDE

However, after updating Platform IO to version 4.4.191002 of the arduinosam package, it now blocks on the AdaFruit SAMD21 boards. I think Platform IO framework is dervied from the AdaFruit one, so it seems likely that one got non-blocking behavior in some older version of the AdaFruit package.

So I'm guessing this behavior is intentional. One could argue that the non-blocking behavior is more portable since that's how Serial.write() works on the three Arduino boards I've tested, as well as the BBC micro:bit, ESP32, and ESP8266. Back when a serial port was just UARTs, it didn't matter if anything was connected to the pin or not; data always got transmitted at the selected baud rate.

Something to think about, anyhow. Meanwhile, now that I understand this issue better MicroBlocks can just ignore the return code from Serial.write on AdaFruit boards and hope for the best...

hathach, 2019-11-07:
Adding either yield() or any sort of delay() to run TinyUSB task, or you can create your own rtos thread just to run the TinyUSB task.

void loop() {
digitalWrite(LED_BUILTIN, HIGH);
**while (!Serial.write(65)) yield()**; // wait until Serial.write() sends a byte
delay(100);
digitalWrite(LED_BUILTIN, LOW);
delay(100);
}

kevinjwalters, 2019-11-07:
FYI, CircuitPython _doesn't_ seem affected by this.

I had this running for 20 minutes on three tests (serial terminal closed shortly after program started; terminal never open; powered by battery with USB dettached). The D13 LED flashing continued on all three. without any cessation of the D13 LED flashing.

```python
import time
import digitalio
import board
boardled = digitalio.DigitalInOut(board.D13)
boardled.direction = digitalio.Direction.OUTPUT

onoff = True

while True:
    print("All work and no play makes Jack a dull boy")
    boardled.value = onoff
    onoff = not onoff
    time.sleep(0.05)
```

jhmaloney, 2019-11-07:
Interesting. Nice to know that CircuitPython isn't affected.

Perhaps this issue is specific to the Arduino Serial.write() implementation?

hathach, 2019-11-08:
thanks for your detail information, I will fix it whenever possible :)

hathach, 2020-09-09:
close since the issue is too old, lots of code is updated. Feel free to re-open again if you sill encounter this with latest code.

closed 2020-09-09 by hathach

jhmaloney, 2020-09-09:
This issue is still extant with the latest Adafruit nrf52 boards package (v0.21.0) for Arduino.

Running the code I supplied on Nov 6, 2019, the LED blinks when the USB cable is plugged in, it does not blink when the board is powered from a battery and is not plugged into a USB cable. The key point is that Serial.write() returns false when the USB cable is not plugged in. I believe it should work the way it works when sending data to a UART pin that isn't connected to anything -- that is, bytes are clocked out through the UART at the baud rate of the serial port making room for additional data in the output buffer.

I can, of course, change my code to ignore the return value of Serial.write(). But that risks transmitting data faster than the serial port hardware can transmit it.

I don't think I'm able to re-open this issue, so I'm hoping you'll see this comment. If not, I suppose I can start a new issue...

hathach, 2020-09-10:
@jhmaloney thanks for testing it again with latest code. Now reading this again, the Serial.write() behavior is indeed correct. Since it isn't able to send out the message, return false is the right and intended behavior. I would suggest you to change your code, either to check for Serial connected, or ignore the return value of write().

jhmaloney, 2020-09-10:
Understood. The Serial.read() behavior has changed over the past year (at least for Adafruit's SAMD21 boards) so it is good to know that the new behavior is intended.

Two questions:

  1. Why doesn't Serial.print() block when the USB cable is not connected? (I'm not saying it should, just wondering.)

  2 .For a Serial port that is uses a USB-serial device, is there a simple way to tell whether it is connected to a host? The ideal solution would be portable across all boards that support the Arduino IDE.

Thanks!

hathach, 2020-09-10:
1. currently Serial.print() check if terminal is connected before putting data into its FIFO, here is its code https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/master/Adafruit_USBD_CDC.cpp#L116
2. You could check if client is connected using bool operator `if (Serial)`, Note that it is to check if there is client connect to the serial e.g IDE monitor. The USB can be already enumerated, you could also check that as well. This is what implemented with TinyUSB library, you got the same with SAMD core if you choose to use TinyUSB stack instead of Arduino one. For other BSP, it totally depends on their implementation that you should check with their author.

https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/master/Adafruit_USBD_CDC.cpp#L72

jhmaloney, 2020-09-10:
Thank you for these super helpful hints!