adafruit/Adafruit_nRF52_Arduino #381: [BUG] warnings building SoftwareSerial example

henrygab, 2019-11-08:
**Describe the bug**
Sample code currently emits warnings.

```
.../SoftwareSerial/SoftwareSerial.cpp:222:6:
    warning: always_inline function might not be inlinable [-Wattributes]

 void SoftwareSerial::recv()
      ^~~~~~~~~~~~~~
```

**Set up (please complete the following information)**
  - Using code from [PR 376](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/376)

**To Reproduce**
Steps to reproduce the behavior:
1. Start a build based on [PR 376](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/376)
2. Allow the build to complete
3. Expand one of the three build jobs that enable additional warnings
4. Review the [build logs](https://travis-ci.com/adafruit/Adafruit_nRF52_Arduino/jobs/254280442)
   for the sample being fixed.

**Expected behavior**
Sample code should build without warnings.

henrygab, 2019-11-08:
OK … I'll admit I'm not entirely sure how I'd remove this warning.

hathach, 2021-06-03:
I am happy to say that this issue is addressed, the recv() is declared as inline but in fact gcc cannot make it inline and have to emit the actual function. I have fixed this recent release, I am glad to say that we are finally building with --all warnings flags in ci https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/tools/build_all.py#L62 . Other Adafruit core and libraries also compile with all in CI now, And you are the big reason to push that. Thank you very much :100: :100:

closed 2021-06-03 by hathach

henrygab, 2021-06-03:
Thank you!  I know you've been extremely busy, and I appreciate the work that went into enabling `--warnings all`.  :)