adafruit/Adafruit_nRF52_Arduino #387: Errors when trying to connect to BLE

MuckYu, 2019-11-10:
I am using a Adafruit NRF52840 Feather Express with the blueart sketch installed.

When I try to connect via the Bluefruit app on Android I get the error "Disconnected, reason = 0x28"
On iOS I get "Disconnected, reason = 0x3E" & "Disconnected, reason = 0x8" in an infinite loop.

Am I doing something wrong or why can I not connect to it?

dhalbert, 2019-11-10:
Could you try with an Apple device as well?

dhalbert, 2019-11-10:
Doing some diagnosis on discord. User has two identical boards: one works, one doesn't. Bootloader has been updated on "bad" one to assure that the SD is there. So this appears to be a hardware issue. Closing for now. Will reopen if necessary.

closed 2019-11-10 by dhalbert