adafruit/Adafruit_nRF52_Arduino #388: Error when compiling tensorflow lite on nrf52

yang12313, 2019-11-10:
**Describe the bug**
Error when compiling tensorflow lite  on nrf52 Arduino environment.
**Set up (please complete the following information)**
  - adafruit bluefruit feather nRF52832
  - Windows 10.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/yang12313/tensorflow_lite_nRF52.git download the model.h and sketch_nov08a.ino
2. please download the associated library which can be found from the library manager
3. Once the setup is done, then click the compile button.
4. error is shown on the readme file.

hathach, 2019-11-11:
seems to be an issue when compiling STL lib, I haven't compiled STL lib with arduino nrf core before.

yang12313, 2019-11-11:
> seems to be an issue when compiling STL lib, I haven't compiled STL lib with arduino nrf core before.
 Thank you so much I am going to unload the STL lib to see if the program can run or not.

bojh, 2019-12-04:
Hi @yang12313 
just done a little test with the hello_world example with nrf52840 Express board. The work around from [here](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/272#issuecomment-510929376) is working. There a still warnings but it does compile and should run (not tested).

hathach, 2020-07-16:
example should work with latest BSP of 0.20.5

closed 2020-07-16 by hathach