adafruit/Adafruit_nRF52_Arduino #390: Serial.write conflict with scan callback

innoagg, 2019-12-05:
When scanning with:

`Bluefruit.begin(0, 1);
Bluefruit.Scanner.setRxCallback(processRawScan);
luefruit.Scanner.clearFilters();
Bluefruit.Scanner.useActiveScan(true);
Bluefruit.Scanner.start(0);`

The callback is fired and the expected payload for the scan and scan response packet is received and the lengths of the packets are as expected. However if inside the callback I write the contents of the payload to Serial with Serial.write() then the payload of the subsequent scan response is empty when it is read in the callback; report->data.len = 0.

In fact any Serial.write call inside the callback will cause the issue. 

`void processRawScan(ble_gap_evt_adv_report_t* report)
{
Serial.write(report->data.p_data, report->data.len);
Bluefruit.Scanner.resume();
}
`
If I write the contents with Serial.print() then the scan response payload remains intact.

BSP Library : 0.14.0
Bootloader  : s140 6.1.1

hathach, 2019-12-09:
can you post your serial output log as well. We have no issue with Serial.write/print in this example. you may want to give it a try https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Central/central_scan/central_scan.ino

innoagg, 2019-12-18:
Sorry for the no reply on this yet. Since I am using the serial output for my app I am having a hard time enabling the debug at the same time without other issues. For example, if I enable debug level 2 I can't do a DFU using VSCode plugin. It hangs when it starts the upload I assume because of the debug output?

Anyway, regarding the original issue with serial.write in the callback, Now I am simply filling a buffer with data and then flushing it out the Serial in the loop(). This resolved it for now.

hathach, 2019-12-19:
@innoagg np at all, we all have little time :)  if you consider this is resolved, would you mind closing this issue.

closed 2019-12-19 by innoagg