adafruit/Adafruit_nRF52_Arduino #391: rssi_proximity_central example fails to compile with VERBOSE_OUTPUT on

oesterle, 2019-12-10:
**Describe the bug**
The Arduino example *File > Examples > Adafruit BlueFruit NRF52 > Projects > rssi_proximity > rssi_proximity_central* won't compile if `VERBOSE_OUTPUT` (line 66) is set to `1`. The compiler complains about multiple undefined constants, including `BLE_GAP_ADV_TYPE_ADV_IND`, which are no longer present in the `ble_gap.h` headers in Nordic SoftDevices. This appeared to break in release **0.9.0**. More info from Adafruit Forums users **ccyclone** and **wbp**: [NRF52 Example Project RSSI Proximity issue](https://forums.adafruit.com/viewtopic.php?f=57&t=157241&p=775211).

**Set up (please complete the following information)**
  - Adafruit Feather nRF52 Bluefruit LE - nRF52832, and BSP v0.14.6
  - macOS 10.14.6
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2):

```
Compiling sketch...
/Users/ericoesterle/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=64000000 -DARDUINO=10808 -DARDUINO_NRF52832_FEATHER -DARDUINO_ARCH_NRF52 "-DARDUINO_BSP_VERSION=\"0.14.6\"" -DNRF52832_XXAA -DNRF52 -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/cmsis/include -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/nrfx -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/nrfx/hal -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/nrfx/mdk -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/nrfx/soc -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/nrfx/drivers/include -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/nrfx/drivers/src -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/nordic/softdevice/s132_nrf52_6.1.1_API/include -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/freertos/Source/include -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/freertos/config -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/freertos/portable/GCC/nrf52 -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/freertos/portable/CMSIS/nrf52 -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/sysview/SEGGER -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/sysview/Config -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/Adafruit_TinyUSB_Core -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5/Adafruit_TinyUSB_Core/tinyusb/src -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/cores/nRF5 -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/variants/feather_nrf52832 -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/Bluefruit52Lib/src -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/SPI -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/Adafruit_LittleFS/src -I/Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/InternalFileSytem/src /var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_build_970200/sketch/rssi_proximity_central.ino.cpp -o /var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_build_970200/sketch/rssi_proximity_central.ino.cpp.o
/var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_modified_sketch_547142/rssi_proximity_central.ino: In function 'void scan_callback(ble_gap_evt_adv_report_t*)':
rssi_proximity_central:248:15: error: 'struct ble_gap_evt_adv_report_t' has no member named 'scan_rsp'
   if (report->scan_rsp)
               ^~~~~~~~
rssi_proximity_central:261:55: error: 'struct ble_gap_evt_adv_report_t' has no member named 'dlen'
   Serial.printf("%14s %d bytes\n", "PAYLOAD", report->dlen);
                                                       ^~~~
rssi_proximity_central:262:15: error: 'struct ble_gap_evt_adv_report_t' has no member named 'dlen'
   if (report->dlen)
               ^~~~
rssi_proximity_central:265:46: error: 'struct ble_gap_evt_adv_report_t' has no member named 'dlen'
     Serial.printBuffer(report->data, report->dlen, '-');
                                              ^~~~
rssi_proximity_central:274:23: error: switch quantity not an integer
   switch (report->type)
                       ^
rssi_proximity_central:276:10: error: 'BLE_GAP_ADV_TYPE_ADV_IND' was not declared in this scope
     case BLE_GAP_ADV_TYPE_ADV_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~
/var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_modified_sketch_547142/rssi_proximity_central.ino:276:10: note: suggested alternative: 'BLE_GAP_AD_TYPE_URI'
     case BLE_GAP_ADV_TYPE_ADV_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~
          BLE_GAP_AD_TYPE_URI
rssi_proximity_central:279:10: error: 'BLE_GAP_ADV_TYPE_ADV_DIRECT_IND' was not declared in this scope
     case BLE_GAP_ADV_TYPE_ADV_DIRECT_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_modified_sketch_547142/rssi_proximity_central.ino:279:10: note: suggested alternative: 'BLE_GAP_AD_TYPE_LE_ROLE'
     case BLE_GAP_ADV_TYPE_ADV_DIRECT_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          BLE_GAP_AD_TYPE_LE_ROLE
rssi_proximity_central:282:10: error: 'BLE_GAP_ADV_TYPE_ADV_SCAN_IND' was not declared in this scope
     case BLE_GAP_ADV_TYPE_ADV_SCAN_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_modified_sketch_547142/rssi_proximity_central.ino:282:10: note: suggested alternative: 'BLE_GAP_AD_TYPE_APPEARANCE'
     case BLE_GAP_ADV_TYPE_ADV_SCAN_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          BLE_GAP_AD_TYPE_APPEARANCE
rssi_proximity_central:285:10: error: 'BLE_GAP_ADV_TYPE_ADV_NONCONN_IND' was not declared in this scope
     case BLE_GAP_ADV_TYPE_ADV_NONCONN_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/folders/14/lvj2khzn1gbfzvjtqg8qg4sh0000gq/T/arduino_modified_sketch_547142/rssi_proximity_central.ino:285:10: note: suggested alternative: 'BLE_GAP_ADDR_TYPE_ANONYMOUS'
     case BLE_GAP_ADV_TYPE_ADV_NONCONN_IND:
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          BLE_GAP_ADDR_TYPE_ANONYMOUS
Using library Bluefruit52Lib at version 0.14.6 in folder: /Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/Bluefruit52Lib 
Using library SPI at version 1.0 in folder: /Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/SPI 
Using library Adafruit_LittleFS at version 0.11.0 in folder: /Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/Adafruit_LittleFS 
Using library InternalFileSytem at version 0.11.0 in folder: /Users/ericoesterle/Library/Arduino15/packages/adafruit/hardware/nrf52/0.14.6/libraries/InternalFileSytem 
exit status 1
'struct ble_gap_evt_adv_report_t' has no member named 'scan_rsp'
```

**To Reproduce**
Steps to reproduce the behavior:
1. In boards, select **Adafruit Bluefruit Feather nRF52832**
2. Go to *File > Examples > Adafruit BlueFruit NRF52 > Projects > rssi_proximity > rssi_proximity_central*
3. Change `VERBOSE_OUTPUT` (line 66) to `1`
4. Verify/Compile the sketch

**Expected behavior**
Compiling with `VERBOSE_OUTPUT` set to `1` should succeed.

wb8wka, 2019-12-22:
any thoughts on this one?

hathach, 2020-01-15:
adv report struct has changed, I have migrated other examples but forgot this since it is not built by default. Thank you very much for reporting the issue.

commit a2013be6 (2020-01-15): fix #391
commit e01ac26d (2020-01-15): Merge pull request #420 from adafruit/develop

fix #391
closed 2020-01-15 by hathach