adafruit/Adafruit_nRF52_Arduino #392: blemidi example won't continue play notes

royno, 2019-12-12:
**Describe the bug**
the blemidi example won't continue playing notes through bluetooh midi. 

**Set up (please complete the following information)**
  - Your Board and BSP version
PCA1059, pca10059_bootloader-0.2.13_s140_6.1.1.hex
  - Your OS (mac/windows/linux) and its version
windows10
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  
cannot check because it happens without serial monitor

**To Reproduce**
Steps to reproduce the behavior:
1. open and upload blemidi example into pca1059 board.
2.  connect it to garageband app of ipad through bluetooth midi protocol.
3. the garageband receives no note from the blemidi example as expected.
4. but if we open a serial monitor (for example, the integrated one of arduino IDE), it will play a note. 
5. click the send button of the serial monitor will play a note once a time. 
6. even closing the monitor will play a note. But it will never continue.

**Expected behavior**
I expect it continue playing note standalone without serial monitor or even USB connected.

royno, 2019-12-15:
I just solved the issue by adding yield(); to end of midiRead() loop. I can't explain why since there is not enough documents on yield() function. If anyone give detailed explanation, I will appreciate very much.

spacenate, 2020-12-11:
I ran in to a different issue with the BLEMIDI example, where BLEMIDI library seemed to be getting stuck in a loop, triggering my `handleNoteOn`/`handleNoteOff` callbacks repeatedly and very fast (even in response to a single note on/off sent to the peripheral). This was somehow getting "reflected"/sent back to the central MIDI host (meaning I was "hearing" the stuck/repeated notes there too). I was not hearing the `note_sequence` notes either (commenting this code out didn't change the behavior).

Adding `yield()` to the end of the `midiRead()` loop as described above resolved this issue for me as well, and now everything works as expected. The comments in [this example code](https://www.arduino.cc/en/Tutorial/MultipleBlinks) suggest the Scheduler requires loops to periodically use `delay()` or `yield()` to prevent one loop from blocking others.

spacenate, 2020-12-13:
Update: upon re-testing, I realized the `yield()` was a false positive for me. The issue I described happens with a specific MIDI app (Xequence for iOS). I will open a separate github issue if I suspect the problem lies with BLEMidi.