adafruit/Adafruit_nRF52_Arduino #405: failure to upload sketch from mac osx

georgekie, 2020-01-01:
**Describe the bug**
A clear and concise description of what the bug is:
Failure to upload sketch to blue feather nrf52840

**Set up (please complete the following information)**
  - Your Board and BSP version
BN: Adafruit Bluefruit Feather nRF52840 Express
VID: 0x239A
PID: 0x8029
SN: F08BE14901382A19

  - Your OS (mac/windows/linux) and its version
10.14.6
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  
ketch uses 101192 bytes (12%) of program storage space. Maximum is 815104 bytes.
Global variables use 14068 bytes (5%) of dynamic memory, leaving 234764 bytes for local variables. Maximum is 248832 bytes.
Upgrading target on /dev/cu.usbmodem14101 with DFU package /private/var/folders/rr/kw3_5dg93tgc9wz1pmblgk_w0000gn/T/arduino_build_214363/sketch_George_Practice.ino.zip. Flow control is disabled, Single bank, Touch disabled
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.
Traceback (most recent call last):
  File "nordicsemi/__main__.py", line 294, in serial
  File "nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
  File "nordicsemi/dfu/dfu.py", line 200, in _dfu_send_image
  File "nordicsemi/dfu/dfu_transport_serial.py", line 179, in send_start_dfu
  File "nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
  File "nordicsemi/dfu/dfu_transport_serial.py", line 282, in get_ack_nr
nordicsemi.exceptions.NordicSemiException: No data received on serial port. Not able to proceed.

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
my email address: georgekieltygmail.com

hathach, 2020-01-07:
Hmm, maybe you have uploaded an sketch that prevent the device to enter DFU properly (e.g `while(1) {}` ). Can you double press the reset button to manually enter DFU mode, and upload again.

lorneb, 2020-03-24:
I can second this problem, a bit more detail I discovered. I am using Mac Osx Arduino Version 1.8.9

Debug level 2 shows
BSP Library : 0.19.0
Bootloader  : s140 6.1.1
Serial No   : 47404FC7C6478D38

if you have a clean board i.e you have just erased it and reloaded bootloader, then you can upload the sample blink sketch, over and over, board enters boot loader automatically  just fine.
As soon as you upload a sketch that uses the softdevice, then that is the last time the board will enter the bootloader by itself.  you have to use the double reset.

If you just reset the user area (I hacked a sketch up) then go back to using the blink sketch then all is well again.

I posted more details in Adafruit forum : [https://forums.adafruit.com/viewtopic.php?f=57&t=163820&p=805003#p805003](url)

hathach, 2020-03-26:
@lorneb thank you for your find, I will further look into the issue, it is probably SD is not happy with a core reset. I will try to reproduce and add code that disable SD properly in the future (currently working  with other project).

lorneb, 2020-03-26:
@hathach ok thanks, if you point me in the direction I don't mind taking a look around.  Or if there are things that I can try in the meantime, I am more than happy to do so.

hathach, 2020-03-26:
Thanks for wiling to try it out, here is the function that got called when touch1200 `enterSerialDfu()`

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/wiring.c#L74

It write the reason to GPRESET register and trigger the core (soft) reset. If I have to do this, I will first check to see if SD is enabled with `sd_softdevice_is_enabled()` like this

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/wiring.c#L97

Then call the `sd_softdevice_disable()` from `nrf_sdm.h` . Though there maybe an more comprehensive shutdown steps for SD. I haven't checked it yet.

lorneb, 2020-03-26:
I'd like to add some more findings( This is more around sketches not starting unless serial present)

it seems that this modification in the latest BSP version 0.19.0
--- Skip waiting for Serial in some examples, force wait for Serial when Debug is enabled
Is what first caused my concern with the sketches not starting automatically unless serial is connected and opened.  
So I have set debug level back to release 0, and carried out the following tests.

1. Load a sketch (blinky) that doesn't use SD, sketch autostarts without serial, Allow next sketch upload without having to double-reset
2. Load a sketch that uses SD, the sketch does NOT auto start and does not allow next sketch to be uploaded, double-reset is required to enter the bootloader.
3. Load a sketch (blinky) that doesn't use SD, sketch autostarts without serial, Allow next sketch upload without having to double-reset

So it seems that autostart without serial (debug set to release) and entering bootloader from the 1200bps touch are somehow related.

lorneb, 2020-03-26:
@hathach  Ok,  I have found an answer:

If your code either via the debug enabled or in the sketch has
`while ( !Serial ) delay(10); `

Then the bootloader is no longer able to be invoked by the 1200bps touch and of course, your sketch does not execute past that point until serial is connected.
Once you remove this from your code then all is well, I am very sorry for the misdirection to the SD being the cause.   This is all tested on mac osx,  have not tested on windows to see if it's even a problem on the Windows platform.

hathach, 2020-03-27:
no these are helpful as well. My reasoning when putting a force wait is, when debug is enabled, user want to connect to Serial for output. Else some of initial output would be missing. I will do more tests with it later on.

lorneb, 2020-03-27:
yep understand the reason behind it now, but for a person not diving in deep it can catch you out as to why the sketch is not running without serial.   I set some breakpoints using gdb in main() and stepped through before I sore the problem,   the gdb test also quickly told me that it was not a bootloader issue.   

So a redefinition of the issue now is:   If you use 
`while ( !Serial ) delay(10);`   or
`while ( !Serial ) yield();`
in your code ( or if it is inserted in your code by debug level > 0 ) then the Arduino bootloader touch 1200 bps method of entering the bootloader no longer works

czuvich, 2020-04-02:
I'd like to chime in... this issue is present even if you omit the Serial calls in the setup() of your sketch.  The only way to run the sketch is to open a Serial Monitor.  I guess I'll try downgrading the BSP?

hathach, 2020-09-09:
closed due to lack of activities

closed 2020-09-09 by hathach