adafruit/Adafruit_nRF52_Arduino #412: nRF52832 Insufficient SRAM for SoftDevice

fyfng, 2020-01-09:
**Describe the bug**
I've previously had working Arduino code on the Adafruit Feather nRF52 Bluefruit LE - nRF52832. I decided to update the BSP Libraries, the Bootloader, and the SoftDevice. Now the code is no longer working. It gives me an error of **App Ram Start must be at least 0x200034A0 (provided 0x20003400)**
I've tried the much simpler examples such as **peripheral/bleuart** (which does not include all of the extra sensors I'm using for my project) and it gives me the same error. I've been struggling with this for a while, so any help would be greatly appreciated.

**Set up**
  - Board: Adafruit Feather nRF52 Bluefruit LE - nRF52832
  - OS: Windows 10
  - Arduino version: 1.8.5
  - Bootloader / SoftDevice: 0.2.13 SoftDevice s132 6.1.1
  - BSP Library: 0.15.1
  - Serial debug log:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Update nRF52832 to Bootloader / SoftDevice 0.2.13 SoftDevice s132 6.1.1 _[Tools > Burn Bootloader]_
2. Open **peripheral/bleuart** example and flash to board _[File > Examples > Adafruit Bluefruit nRF52 Libraries > Peripheral > bleuart]_
3. Open _[Tools > Serial Monitor]_
4. See error **App Ram Start must be at least 0x200034A0 (provided 0x20003400)**

**Expected behavior**
The default example should work with the latest Bootloader and SoftDevice.

**Additional context**
I believe the previous BSP Library I was using that was not giving me any issues was 0.8.3 with a SoftDevice of S132 2.0.1 (I think). I've tried downgrading to BSP Library 0.11.1 and BSP Library 0.14.6 and both of them gave me the same issues.

I've consulted this page: https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/hathach-memory-map and tried changing some functions to reduce the SRAM usage by the SoftDevice.

- I've tried reducing the bandwidth for the **configPrphBandwidth()** function to BANDWIDTH_LOW, but it was not able to sufficiently reduce the SRAM of the SoftDevice.
- I've tried setting the parameters for the **begin()** function to (1,0), but it was not able to sufficiently reduce the SRAM of the SoftDevice.
- I've tried modifying the linker located at: **AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.15.1\cores\nRF5\linker\nrf52832_s132_v6.ld** by changing the RAM Origin and Length by changing 0x20003400 to 0x200034A0 or 0x200034A1, but both times still gave me the same error. Maybe I did not revise it properly.

mattratto, 2020-01-10:
Just FYI I resolved this issue by editing /cores/nRF5/linker/nrf52832_s132_v6.ld: 

  (line 17) RAM (rwx) :  ORIGIN = 0x200034A0, LENGTH = 0x20010000 - 0x200034A0

note that I changed both the origin and the length.

hathach, 2020-01-15:
thank you for reporting the issue, I could reproduce and have fixed this. We are more focusing on 52840 and doesn't perform enough testing. That is why we need more user feedback like yours. We will include this fix in the next release. RAM for SD is increased to 0x3600 which provide a bit more head room for other users as well.

commit 9bd542e3 (2020-01-15): increase SD ram for nrf52832

- from 0x3400 to 0x3600, fix #412
PR #418 (2020-01-15): increase SD ram for nrf52832
- from 0x3400 to 0x3600, fix #412
closed 2020-01-15 by hathach