adafruit/Adafruit_nRF52_Arduino #419: Ambigous definition of File between Adafruit_LittleFS and SD library

Protonerd, 2020-01-15:
Definition of File is ambigous and causes a compile error if both SD library and Adafruit_LittleFS are used, see error message from Arduino IDE below.

I'm using an Adafruit BluFeather BLE with an nRF52, Bootloader 0.2.11 Softdevice s132 6.1.1
Arduino IDE version 1.8.9

The sketch I wrote contains references to the LittleFS as well as using the SD lib.
#include <Adafruit_LittleFS.h>
#include <SD.h>

If I change in the Adafruit_LittleFS the name File to for instance LFS_File, then the error disappers, and my sketch compiles. Problem is, that in this case other sketches from other Adafruit sketches which expects File will not compile.

```
Sound.h:28:21: error: reference to 'File' is ambiguous

In file included from sketch\Sound.h:7:0,

                 from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\nRF52_NordicSnippets_trials\nRF52_NordicSnippets_trials.ino:6:

D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\libraries\SD\src/SD.h:28:7: note: candidates are: class SDLib::File

 class File : public Stream {

       ^~~~

In file included from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\libraries\Adafruit_LittleFS\src/Adafruit_LittleFS.h:34:0,

                 from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\libraries\Infinity_Basics/Infinity_Basics.h:29,

                 from sketch\Button.h:8,

                 from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\nRF52_NordicSnippets_trials\nRF52_NordicSnippets_trials.ino:5:
```

hathach, 2020-01-15:
could you provide a simple sketch that we can use to reproduce the issue.

Protonerd, 2020-02-05:
Sorry hathach for the delayed response, now I could find the time to revert my quick fix changes and make a really easy sketch (actually more or less an empty Arduino one) to replicate the problem:

`#include <SPI.h>
#include <SD.h>
#include <Adafruit_LittleFS.h>
#include <InternalFileSystem.h>
using namespace Adafruit_LittleFS_Namespace;

File file(InternalFS);
File myFile;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

And here is the error message from the IDE:
SD_LFS_Trial:9:1: error: reference to 'File' is ambiguous

 File file(InternalFS);

 ^~~~

In file included from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\libraries\Adafruit_LittleFS\src/Adafruit_LittleFS.h:33:0,

                 from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\SD_LFS_Trial\SD_LFS_Trial.ino:5:

D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\libraries\Adafruit_LittleFS\src/Adafruit_LittleFS_File.h:41:7: note: candidates are: class Adafruit_LittleFS_Namespace::File

 class File : public Stream

       ^~~~

In file included from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\SD_LFS_Trial\SD_LFS_Trial.ino:4:0:

D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\libraries\SD\src/SD.h:28:7: note:                 class SDLib::File

 class File : public Stream {

       ^~~~

SD_LFS_Trial:10:1: error: reference to 'File' is ambiguous

 File myFile;

 ^~~~

In file included from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\libraries\Adafruit_LittleFS\src/Adafruit_LittleFS.h:33:0,

                 from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\SD_LFS_Trial\SD_LFS_Trial.ino:5:

D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\libraries\Adafruit_LittleFS\src/Adafruit_LittleFS_File.h:41:7: note: candidates are: class Adafruit_LittleFS_Namespace::File

 class File : public Stream

       ^~~~

In file included from D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\portable\sketchbook\SD_LFS_Trial\SD_LFS_Trial.ino:4:0:

D:\Arduino\nRF52_Infinity_DevEnv\arduino-1.8.9\libraries\SD\src/SD.h:28:7: note:                 class SDLib::File

 class File : public Stream {

       ^~~~

exit status 1
reference to 'File' is ambiguous

Thanks a lot for lokking into this topic and for all your amazing support of the community!
Protonerd

Protonerd, 2020-02-20:
Is there any feedback on this issue? I expect it seriously impair the usability of the library if it clashes with the widely used SD library...

hathach, 2020-02-20:
I am overwhelmed with works, and couldn't look at this issue for now. Maybe you should try to use the namespace explicit instead of `using` e.g `Adafruit_LittleFS_Namespace::File` throughout your sketches. typedef or define it could make it shorter.

hathach, 2020-03-04:
Have you try the explicit namespace, I guess it should work

Protonerd, 2020-03-04:
I’m not sure what you mean by explicit namespace...

hathach, 2020-03-04:
Replace File that is intended for internal fs with Adafruit_LittleFS_Namespace::File in your sketch

Protonerd, 2020-03-04:
Using my simple demo sketch this indeed got rid of the compile error

Here is the modified sketch for reference:
`#include <SPI.h>
#include <SD.h>
#include <Adafruit_LittleFS.h>
#include <InternalFileSystem.h>

Adafruit_LittleFS_Namespace::File file(InternalFS);
File myFile;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

Thanks hathach, I will go on testing it with a more sophisticated sketch.

hathach, 2020-03-05:
@Protonerd Glad you get it work, i think it should be fine, it is the File naming conflict. C++ wrap name inside an namespace, `using namespace abc` allow us to use the name without its namespace prefix for cleaner look. Since there is conflict, we must explicitly tell which space File is used. 

Tip: you could `#define iFile  Adafruit_LittleFS_Namespace::File` or intFile etc.. to make it look more clean in your sketch. Let's me know if you have other issue with it.

henrygab, 2020-03-06:
Hi @Protonerd,

@hathach described it correctly; here is a longer description, with links to even more information.

<details><summary>Namespace background</summary>

</P>

First, it is important to understand that C++ has the concept of [namespaces](https://en.cppreference.com/w/cpp/language/namespace).  A namespace is simply a way to prevent naming conflicts.  Similar to namespaces in other languages, when something is defined in a namespace, it is no longer visible in the global namespace.

**_By default_**, to use a thing defined in a namespace, you have to prefix the name with the namespace and two colons.  You've probably already seen this.  For example, [`std::cout << "hello world\n"`](https://en.cppreference.com/w/cpp/io/cout) uses the object `cout` that is defined in the namespace `std`.
</details>
<details><summary>Problem caused by the `using` line</summary>

</P>
```C++
using namespace Adafruit_LittleFS_Namespace;
```
This brings every object that was defined in the namespace `Adafruit_LittleFS` into the global namespace, including the object `Adafruit_LittleFS::File`.  The problem is, the SD library defined their own object named 'File' in the global namespace.  Thus, because the `using` statement modified how names are looked up, the compiler cannot know which `File` object type your were intending to refer to.
</details>
<details><summary>Elegant solution</summary>

</P>
The brute-force method to solve this is to remove your `using` statement, and then fully-qualify all access to the Adafruit_LittleFS objects using the namespace prefix.  But, that's cumbersome.

A more elegant solution is to use a slightly different form of the `using` line called a which only pulls in a single named item from the namespace.  

The following two lines have nearly identical results, but the second line is more elegant by applying a C++ [using declaration](https://en.cppreference.com/w/cpp/language/namespace#Using-declarations):
```C++
// #define iFile Adafruit_LittleFS_Namespace::File
using iFile Adafruit_LittleFS_Namespace::File;
```

Why is the second line more elegant?  The second line can be used in ways that do not pollute the global namespace.  But, that's a deeper discussion -- look at [name lookup](https://en.cppreference.com/w/cpp/language/lookup) and the page on [namespaces](https://en.cppreference.com/w/cpp/language/namespace) (_hint: can be used inside your own namespace, or even inside a single block, without affecting stuff outside that scope_).

</details>

As a summary, replace your **_using declaration_**:
```C++
using namespace Adafruit_LittleFS_Namespace;
```

With the following **_using directive_**:
```C++
using LfsFile Adafruit_LittleFS_Namespace::File;
```

And then use the name `LfsFile` ... without having to change the InternalFS library at all.

Protonerd, 2020-03-12:
Great explanation henrygab, thanks for educating me on this topic! Today was not a wasted day at all :) I close this issue grateful for your support guys!

closed 2020-03-12 by Protonerd