adafruit/Adafruit_nRF52_Arduino #422: Adafruit_nRF52_Arduino-master\libraries\Bluefruit52Lib\examples\Peripheral\bleuart\ failed to start "radio"

eduards66, 2020-01-16:
running Blueuart.uno for peripheral failed properly initialize nrf52832 feather and then there is no blue LED  that indicates radio working.
The reason is attempt to Config the peripheral connection with maximum bandwidth (line 40):
  Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);
The code works fine if this line is commented out.
SET up: 
- arduino ide 1.8.11 hourly build 2020/01/08 12:33
- Board: adafruit feather nRf52832, 
   bootloader 0.2.13 SoftDevice s132 6.1.1. Level0(Release)
- windows10
 
**To Reproduce**
1.Load \Adafruit_nRF52_Arduino-master\libraries\Bluefruit52Lib\examples\Central\central_bleuart\ to central device(works as expected: blue LED is ON to indicate "RADIO" ON)
2. Load Adafruit_nRF52_Arduino-master\libraries\Bluefruit52Lib\examples\Peripheral\bleuart\  to peripheral and blue LED will stay OFF forever.
3. The FIX is to comment out line 40 : Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);

NOTE: (This bug is related to the latest code from: https://github.com/adafruit/Adafruit_nRF52_Arduino at 2020/01/16) 
- I do not know the actual reason for failure, but without that line board works as expected.
- The other example bleuart_multi does not have this line and it works as is.

hathach, 2020-01-22:
Thanks for reporting, please run the sketch with DEBUG  level set to 1 or 2 using IDE menu selection, and paste your serial output here. It is possibly the same issue with this https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/412 , which is fixed recently.

eduards66, 2020-01-22:
I added the changes from github to: ...\packages\adafruit\hardware\nrf52\0.16.0\cores\nRF5\linker\nrf52832_s132_v6.ld and radio started...
Thanks a lot :-)

hathach, 2020-01-23:
@eduards66 thanks for the issue, Please also name the github issue with meaningful subject in the future

closed 2020-01-23 by hathach