adafruit/Adafruit_nRF52_Arduino #424: adafruit-nrfutil.exe is deleted from ...\\arduino-nightly\\portable\\packages\\adafruit\\hardware\\nrf52\\0.16.0/tools/adafruit-nrfutil/win32/ after successful build.

eduards66, 2020-01-17:
**Describe the bug**
adafruit-nrfutil.exe is deleted from ...\\arduino-nightly\\portable\\packages\\adafruit\\hardware\\nrf52\\0.16.0/tools/adafruit-nrfutil/win32/ after successful build and then next time build will fail on

**Set up (please complete the following information)**
SET up:

arduino ide 1.8.11 hourly build 2020/01/08 12:33
Board: adafruit feather nRf52832,
bootloader 0.2.13 SoftDevice s132 6.1.1. Level0(Release)
windows10

**To Reproduce**
Steps to reproduce the behavior:
1. build the \Adafruit_nRF52_Arduino-master\libraries\Bluefruit52Lib\examples\Central\central_bleuart\
2. Verify that it successfully done.
3. Rebuild it again.
4. See error, it can be clear error like adafruit-nrfutil.exe does not exist or any other that you can find in the attached LOG file.
5. check \\arduino-nightly\\portable\\packages\\adafruit\\hardware\\nrf52\\0.16.0/tools/adafruit-nrfutil/win32/ for adafruit-nrfutil.exe. In case of error this file does not exist.
6. To resolve: copy adafruit-nrfutil.exe to the proper location:
.\\arduino-nightly\\portable\\packages\\adafruit\\hardware\\nrf52\\0.16.0/tools/adafruit-nrfutil/win32/ and build it again and see the successful result.
[err_log.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/4074418/err_log.txt)
[success_log.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/4074429/success_log.txt)
NOTES:
- in ARDUINO IDE selected Programmer: "Bootloader DFU for Bluefruit nrf52"
- sometimes it take a couple of times to reproduce the issue but it happen consistently. 

**Additional context**
Add any other context about the problem here.

eduards66, 2020-01-17:
[err_log_1.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/4074509/err_log_1.txt)

exec: "\\arduino-nightly\\portable\\packages\\adafruit\\hardware\\nrf52\\0.16.0/tools/adafruit-nrfutil/win32/adafruit-nrfutil.exe": file does not exist
Error compiling for board Adafruit Bluefruit Feather nRF52832.

hathach, 2020-01-17:
please check to see if the exe file is blocked by your anti-virus, there is a previous report that McAfee block nrfutil

eduards66, 2020-01-20:
No, McAfee does not block it...

eduards66, 2020-01-20:
Sorry, accidentally closed it, then reopen again...

hathach, 2020-01-22:
adafruit-nrfutil existed but then removed from your system, please check to see which process delete it and why. It is possible that your pc mis-identify it as malicious ware. I don't know what we can do to resolve this.

eduards66, 2020-01-22:
it is deleted after every single build, then it is not related to background pc activity.

hathach, 2020-01-23:
And have you figured out why and what remove it ?

eduards66, 2020-01-24:
1. downloaded  Adafruit_nRF52_Arduino-master from https://github.com/adafruit/Adafruit_nRF52_Arduino
2. copied its content to ...\nrf52\0.16.0\
3.  substituted  adafruit-nrfutil.exe with the latest from https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases
=== Can not reproduce BUG anymore ===

hathach, 2020-01-30:
I am glad you work it out.

closed 2020-01-30 by hathach