adafruit/Adafruit_nRF52_Arduino #425: Feather nRF52840: OTA BLE fails

rei-vilo, 2020-01-18:
**Describe the bug**
I'm trying to perform a BLE OTA upload to the Feather nRF52840.

Adafruit Bluefruit LE Connect lists and connects to the Feather nRF52840 but fails to upload an HEX fails with the error message

> The legacy bootloader on this device is not compatible with this application.

**Set up (please complete the following information)**
The configuration includes:

- UF2 Bootloader 0.2.13 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-755-g55874813) s140 6.1.1
- Model: Adafruit Feather nRF52840 Express
- Board-ID: nRF52840-Feather-revD
- Date: Sep 25 2019
- Adafruit Bluefruit LE Connect: Version 2.2.2 (22
- Sketch: dfu_ota.ino

**To Reproduce**
Steps to reproduce the behavior:
1. Adafruit Bluefruit LE Connect lists and connects to the Feather nRF52840.
[image]

2. However, trying to upload an HEX fails with the error message
[image]

**Expected behavior**
Perform OTA upload.

**Screenshots**
See above.

**Additional context**

I went through different forum threads ([1](https://forums.adafruit.com/viewtopic.php?f=57&t=123583&hilit=nrf52840+OTA), [2](https://forums.adafruit.com/viewtopic.php?f=19&t=159272&p=784468&hilit=nrf52840+OTA), [3](https://forums.adafruit.com/viewtopic.php?f=57&t=123583&hilit=otal)) and repository issues ([4](https://github.com/charlesportwoodii/kaidyth_nrf52_bootloader/issues/3),[ 5](https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/35)), to no avail. 

How to proceed to enable BLE OTA and keeping UF2 feature? 

Adding a dedicated tutorial on the Learn section would be great.

Thank you!

hathach, 2020-01-20:
the macos application doesn't support nrf52, you can perform OTA with nrftoolbox & nrfconnect application for ios or android

closed 2020-01-20 by hathach

rei-vilo, 2020-01-20:
Thanks for the clarification.

Updating the related [Introducing the Adafruit nRF52840 Feather](https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/overview) section and providing a working OTA procedure would be great.

hathach, 2020-01-22:
Yeah, things evolving, code and docs can be out of sync. We will work on that later on.