adafruit/Adafruit_nRF52_Arduino #428: Add support for sdk_config.h

binloan, 2020-01-24:
**Is your feature request related to a problem? Please describe.**

While looking through the sources, I noticed that the NRF SDK is not full featured, but uses only a small subset - which is totally fine.
However, I also noticed that if I wanted to add some additional modules such as the `crypto` module from the NRF SDK, it would require me to satisfy some additional dependencies.
In special the `sdk_config.h`. I know this file is normally used to configure Nordics SDK features, but it also configures some features such as the clock source, softdevice parameters and so on. I wonder why you decided to get rid of the `sdk_config.h`?

The current approach feels very intransparent and, in my opinion, will lead to issues with configurations, especially if someone decides that he requires a NRF feature that isn't available through the current implementation. 

**Describe the solution you'd like**
Add the `sdk_config.h` to allow extending the code base for more advanced projects.
Document the locations where `sdk_config.h` values are overwritten to prevent confusion and issues. 

**Describe alternatives you've considered**
Have both the `sdk_config.h` and the inline configuration of the SDK. However, this will create a lot of change overhead whenever a different feature is required.

**Additional context**
None.

Thanks!

hathach, 2020-01-30:
this repo doesn't make use of sdk sources, we only use nrfx for peripheral drivers

closed 2020-01-30 by hathach

binloan, 2020-01-30:
@hathach Just because I'm curious - is there a reason why you chose to do this?

hathach, 2020-02-03:
SDK License was used to be very restricted and not appropriate for opensource project. We had to find our way around that, also sdk is too heavy and constantly changing. At this stage, I don't see the need to use SDK anymore.

ladyada, 2020-02-03:
SDK is not good to rely on, even nordic recommended we use nrfx instead!

binloan, 2020-02-14:
@ladyada I just got to hear that from some nordic engineers as well! They seem to be very convinced of their SDKs.

@hathach ok sounds reasonable. I was not aware of the licences. Thanks for clearing this up!