adafruit/Adafruit_nRF52_Arduino #433: Particle Xenon - Issue with Onboard LED's

DirtyEngineer, 2020-02-03:
With the recent news from Particle regarding the Xenon I have migrated to using the Adafruit bootloader as they suggested by Particle. Exploring this package I found that it does indeed have the variant framework to support programming via the Ardunio IDE (Particle Xenon support #317).  All that was needed is to add it to the board.txt file.

```
# ----------------------------------
# Particle Xenon nRF52840
# ----------------------------------
xenon52840.name=Particle Xenon

# VID/PID for bootloader with/without UF2, Arduino + Circuitpython App
xenon52840.vid.0=0x239A
xenon52840.pid.0=0x8029
xenon52840.vid.1=0x239A
xenon52840.pid.1=0x0029
xenon52840.vid.2=0x239A
xenon52840.pid.2=0x002A
xenon52840.vid.3=0x239A
xenon52840.pid.3=0x802A

# Upload
xenon52840.bootloader.tool=bootburn
xenon52840.upload.tool=nrfutil
xenon52840.upload.protocol=nrfutil
xenon52840.upload.use_1200bps_touch=true
xenon52840.upload.wait_for_upload_port=true
xenon52840.upload.maximum_size=815104
xenon52840.upload.maximum_data_size=237568

# Build
xenon52840.build.mcu=cortex-m4
xenon52840.build.f_cpu=64000000
xenon52840.build.board=PARTICLE_XENON
xenon52840.build.core=nRF5
xenon52840.build.variant=particle_xenon
xenon52840.build.usb_manufacturer="Particle Industries"
xenon52840.build.usb_product="Particle Xenon"
xenon52840.build.extra_flags=-DNRF52840_XXAA {build.flags.usb}
xenon52840.build.ldscript=nrf52840_s140_v6.ld
xenon52840.build.vid=0x239A
xenon52840.build.pid=0x8029

# SofDevice Menu
xenon52840.menu.softdevice.s140v6=0.2.13 SoftDevice s140 6.1.1
xenon52840.menu.softdevice.s140v6.build.sd_name=s140
xenon52840.menu.softdevice.s140v6.build.sd_version=6.1.1
xenon52840.menu.softdevice.s140v6.build.sd_fwid=0x00B6

# Debug Menu
xenon52840.menu.debug.l0=Level 0 (Release)
xenon52840.menu.debug.l0.build.debug_flags=-DCFG_DEBUG=0
xenon52840.menu.debug.l1=Level 1 (Error Message)
xenon52840.menu.debug.l1.build.debug_flags=-DCFG_DEBUG=1
xenon52840.menu.debug.l2=Level 2 (Full Debug)
xenon52840.menu.debug.l2.build.debug_flags=-DCFG_DEBUG=2
xenon52840.menu.debug.l3=Level 3 (Segger SystemView)
xenon52840.menu.debug.l3.build.debug_flags=-DCFG_DEBUG=3

```
[boards.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/4146209/boards.txt)

Sweet, the board programs without issue. This is great!  The problem is that there's an issue with accessing the onboard LED's. From my understanding the way to access them is with;

- LED_BUILTIN
- LED_RGB_RED
- LED_RGB_GREEN
- LED_RGB_BLUE

This should control these pins respectively;

- P1.12
- P0.13
- P0.14
- P0.15

The blink example will compile but does not toggle any of the LEDS's using the LED_XXXXX terms above.  I can get the built in LED to toggle by using the blink example with Digital pin 7 as they use the same pin but I can't seem to find a way to access the RGB LED's on the board.

hathach, 2020-02-03:
I am glad to hear you could get xenon working with our repo, however we don't provide detailed technical support for boards that are not made by Adafruit. Though we are more than happy to accept PRs from Particle user.

DirtyEngineer, 2020-02-03:
It's not technical support, it's an issue with code that you accepted a pull request for.

ladyada, 2020-02-03:
@DirtyEngineer hi please submit a fix PR and we'll take a look!

DirtyEngineer, 2020-02-03:
I'm trying to figure it out. Maybe @outlandnish from the original Particle Xenon support #317 can take a look as well.

outlandnish, 2020-02-03:
I've got a fix for it, but I hadn't done a PR for it. I'll push the changes later today

outlandnish, 2020-02-05:
Created a PR (https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/435) to add more intuitive ways of controlling the LEDs (as well as the boards.txt file)

DirtyEngineer, 2020-02-05:
@outlandnish  Thank you!

closed 2020-02-05 by DirtyEngineer