adafruit/Adafruit_nRF52_Arduino #438: 'Upload using programmer' fails when using local clone of BSP

henrygab, 2020-02-11:
**Describe the bug**
When BSP is local, users cannot use 'Upload using programmer' to flash.

**Set up (please complete the following information)**
  - Board == AdaFruit Feather nrf52840 Express
  - BSP == commit e932935cab6f2103b5cba1b36ccbced92110029a (0.18.0), but also occurred on prior official released (by sync'ing to release version tags)
  - OS == Windows
  - Serial debug log == N/A … occurs at compilation

**To Reproduce**
Steps to reproduce the behavior:
1. Set sketch directory to some location  
`D:\SRC\Sketches`
2. Make the hardware\AdaFruit subdirectories to store the BSP  
`md D:\src\Sketches\hardware\Adafruit`
3. Change to that directory  
`cd D:\src\Sketches\hardware\Adafruit``
3. Clone the BSP (this depot) to `nrf52` subdirectory  
`git clone https://github.com/adafruit/Adafruit_nRF52_Arduino.git nrf52`
4. Pull v0.18.0 sources  
`git pull`
`git submodule init`
`git checkout 0.18.0`
`git submodule update`
5. Start Arduino IDE
6. In Arduino IDE...
    a. Set `AdaFruit Feather nrf52840 Express` as board
    b. Load the 'Blink' example sketch
    c. Build / program the 'Blink' example sketch over COM port (works)
    d. Connect JLink to the AdaFruit Feather nrf52840 Express board
    e. Choose 'Upload using Programmer'

**Expected behavior**
The upload should work.

**Actual behavior**
* Build succeeds
* Linking succeeds
* `arm-none-eabi-size` succeeds, showing size of the binary
* Error message is printed to console in red:
```
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:295)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:90)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:748)
```

**Additional context**
Uploading using the JLINK works properly when the BSP is installed in the normal hardware subdirectory.  However, this conflicts with the board manager, and requires the name of the subdirectory containing the files to continually be updated to reflect the version number... in other words, cloning the repo in the same place as the board manager would place the files is neither supported nor recommended.  The correct location for a BSP during development is under the sketch folder, under the `hardware` subdirectory.

This should be a relatively easy fix, once the cause is nailed down.  It might just be that the `nrfjprog.exe` needs to be part of this depot, or that some reference to the tool needs to be updated to point to a different path that updates correctly when the BSP is in a different location?

hathach, 2020-03-09:
I can't reproduce this issue, I am always using the git clone BSP to Arduino/hardware/adafruit/nrf52 and haven't got any issue with uploading. There is `adafruit-nrfutil.exe` included with bsp https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/tools/adafruit-nrfutil/win32, sometime it got removed by anti-virus program.please check your local folder to see if it exists.

Also please post more of your compile log, that also includes the upload command `adafruit-util.exe ....` as well.

henrygab, 2020-03-10:
Interesting.  Just to confirm:
1. Did you place the `hardware\adafruit\nrf52\` directory under the user's sketch folder?
2. Are you using JLINK programmer (see step 7d)?

I also had no issue when I cloned to `%LOCALAPPDATA%\Arduino15\packages\adafruit\hardware`.
However, when I cloned to the user's sketch folder, this problem occurred.

I confirm the adafruit-nrfutil.exe program is available and runs manually.  I will get you as full a log as possible shortly (currently in midst of larger effort)

hathach, 2020-03-11:
@henrygab I placed it into the user sketch folder (along with the libraries). You may want to have an extra version folder as well e.g `hardware\adafruit\0.50.0\nrf52` it seems to be the new way Arduino IDE use to keep multiple version of BSP. However I only test it on Linux, you don't have to spend too much time on this. The longer compile log is all I need, I will try to reproduce it with an windows machine later, btw are you using windows 7 or 10 ?

henrygab, 2020-03-11:
@hathach -- **_Thank you_** for looking at this.  

I can no longer reproduce this issue. Perhaps there was some interim issue with a Windows Insider build?  (I have now left that program.)  Or, perhaps it was AV interference?  Or, perhaps I simply made a mistake.  Regardless of the cause, I will now close this.  **_Thank you again_** for looking!

closed 2020-03-11 by henrygab

hathach, 2020-03-12:
> @hathach -- **_Thank you_** for looking at this.
>
> I can no longer reproduce this issue. Perhaps there was some interim issue with a Windows Insider build? (I have now left that program.) Or, perhaps [...]

probably the work of an AV program, it has been reported that some will quarantine the `adafruit-nrfutil.exe`. Thank you and feel free to open an issue when possible