adafruit/Adafruit_nRF52_Arduino #450: Arduino IDE claims 0.18.0 can be updated, but no newer version available

dhalbert, 2020-03-09:
I'm getting an updatd notice from Linux Arduino IDE 1.8.12 that this BSP needs to be updated, but I have 0.18.0 installed, and there's no newer version. So maybe there's a version skew somewhere?

colindgrant, 2020-03-11:
In case it's useful to have as a datapoint:
Arduino 1.8.11 on Mac Mojave allowed me to fresh-install 0.18.5 today:

[image]

hathach, 2020-03-12:
thanks @colindgrant for confirmation, @dhalbert that is weird, there should be an 0.18.5, which OS you are on !? I will check it out.

dhalbert, 2020-03-12:
This is on Ubuntu 20.04 (but I was having the same trouble on 18.04). The BSP is listed as updatable, but the new version isn't visible. Like @colindgrant, I installed the BSP on a new installation of Arduino on another Ubuntu 20.04 computer, and it did install 0.18.5

[image]

hathach, 2020-03-12:
@dhalbert this is headache, I have no clues atm. I just release 0.19.0 https://github.com/adafruit/arduino-board-index/pull/61 . Let's wait until tomorrow to see if it is updated

colindgrant, 2020-04-28:
One last datapoint in case it's useful: I was able to install 0.19.0 and now I'm on 0.20.1, both installed via Arduino IDE with no apparent issue. MacOS Mojave/10.14.6, Arduino 1.8.12.

hathach, 2020-07-14:
I just upgraded my PC to ubuntu 20.04, everything seems to be fine now. This should be closed, can be reopened if anyone still has issue with it.

closed 2020-07-14 by hathach