adafruit/Adafruit_nRF52_Arduino #456: I2C Wire write conflicting with Print write

caternuson, 2020-03-12:
BSP 0.18.5
Board = Feather nRF52832 (but same with others)

**NOTE**: it will compile with an explicit cast `Wire.write(uint8_t(0x00));`

```cpp
#import <Wire.h>

void setup() {
  Wire.begin();
  Wire.beginTransmission(0x50);
  Wire.write(0x00);
  Wire.endTransmission();
}

void loop() {
}
```

```
/home/test/Arduino/nrf52832_i2c_check/nrf52832_i2c_check.ino: In function 'void setup()':
nrf52832_i2c_check:6:18: error: call of overloaded 'write(int)' is ambiguous
   Wire.write(0x00);
                  ^
In file included from /home/test/Arduino/nrf52832_i2c_check/nrf52832_i2c_check.ino:1:0:
/home/test/.arduino15/packages/adafruit/hardware/nrf52/0.18.5/libraries/Wire/Wire.h:50:12: note: candidate: virtual size_t TwoWire::write(uint8_t)
     size_t write(uint8_t data);
            ^~~~~
In file included from /home/test/.arduino15/packages/adafruit/hardware/nrf52/0.18.5/cores/nRF5/Stream.h:26:0,
                 from /home/test/.arduino15/packages/adafruit/hardware/nrf52/0.18.5/cores/nRF5/HardwareSerial.h:25,
                 from /home/test/.arduino15/packages/adafruit/hardware/nrf52/0.18.5/cores/nRF5/Arduino.h:55,
                 from /tmp/arduino_build_600571/sketch/nrf52832_i2c_check.ino.cpp:1:
/home/test/.arduino15/packages/adafruit/hardware/nrf52/0.18.5/cores/nRF5/Print.h:51:12: note: candidate: size_t Print::write(const char*)
     size_t write(const char *str) {
            ^~~~~
Using library Wire at version 1.0 in folder: /home/test/.arduino15/packages/adafruit/hardware/nrf52/0.18.5/libraries/Wire 
exit status 1
call of overloaded 'write(int)' is ambiguous
```

edgar-bonet, 2020-03-12:
I took a look at the various overloads of `Wire.write()`, comparing the cores [arduino/ArduinoCore-avr][core-avr] and [adafruit/Adafruit\_nRF52\_Arduino][core-nrf]. What I found is:

 * Print.h defines the same overloads on both cores
 * for Wire.h, the AVR core defines some extra overloads not present on the nRF52 core.

[These extra overloads][overloads] are:

```c++
inline size_t write(unsigned long n) { return write((uint8_t)n); }
inline size_t write(long n)          { return write((uint8_t)n); }
inline size_t write(unsigned int n)  { return write((uint8_t)n); }
inline size_t write(int n)           { return write((uint8_t)n); }
```

It would seem this is what makes the overload resolution unambiguous on AVR. I suggest adding these overloads to this core.

[core-avr]: /arduino/ArduinoCore-avr
[core-nrf]: /adafruit/Adafruit_nRF52_Arduino
[overloads]: /arduino/ArduinoCore-avr/blob/1.8.2/libraries/Wire/src/Wire.h#L75-L78

hathach, 2020-03-13:
thanks @edgar-bonet for detailed analysis, adding those inline indeed help to resolve the ambiguous calls. I have the submit PR #456, it will fix this when merged to master.

closed 2020-03-13 by hathach

caternuson, 2020-03-13:
Nice! Thanks @edgar-bonet and @hathach

colindgrant, 2020-04-18:
Hi @hathach,

I hope I'm not imposing, but I don't think that this change made it into the 0.19.0 nRF52 BSP as I've seen referenced [here](https://github.com/adafruit/RTClib/issues/136#issuecomment-603217664) and [here](https://forums.adafruit.com/viewtopic.php?f=25&t=163404&p=808477#p803341), and I think there may actually be a change that is prudent (if not specifically required) in the RTCLib repo as well.

I posted details in the forum where this issue was initially brought up:
https://forums.adafruit.com/viewtopic.php?f=25&t=163404&p=808477#p808477

CC: @caternuson @edgar-bonet @henrygab @drak7

I welcome feedback from any of you, I'm happy to tee up a PR or help any way I can, and I hope everyone is staying safe!

edgar-bonet, 2020-04-19:
@colindgrant: The fix you are proposing (casting to `byte` the argument of `Wire._I2C_WRITE()`) is something I would call a “workaround”: a common approach when dealing with a bug in a library that you cannot fix and whose authors are not very responsive.

In this case, however, both the nRF52 core and RTClib are maintained by the same team. I do not think it would make much sense to them to implement in RTClib a workaround for an issue in Adafruit\_nRF52\_Arduino. Especially considering that the issue is fixed in master and needs only to be released.

In the mean time, a reasonable user-level workaround would be to install Adafruit\_nRF52\_Arduino from master (or any commit in master from March&nbsp;13 onwards) rather than 0.19.0.

colindgrant, 2020-04-19:
@edgar-bonet ok, great. Thank you for the extra review and the explanation. @henrygab thanks for your forum reply too, now others who find that post will know to look for the fix in the next release. 👍

commit a2d32e97 (2020-03-13): fix #456 Wire Write ambiguous
commit be74f529 (2020-03-13): Merge pull request #457 from adafruit/fix-wire-write

fix #456 Wire Write ambiguous