adafruit/Adafruit_nRF52_Arduino #458: analogWriteResolution() does not update PWM3 peripheral

henrygab, 2020-03-13:
**Describe the bug**
Function `analogWriteResolution()` updates each of `HwPWM0`, `HwPWM1`, and `HwPWM2`.  However, the nRF52840 has a fourth hardware PWM peripheral, `HwPWM3`, and while that fourth peripheral is used by analogWrite() (when needed), this peripheral is never updated by `analogWriteResolution()`

**Set up (please complete the following information)**
Discovered via code review while investigating #447.  Applies to all nRF52840 boards.

Expected repro steps:
Call analogWrite() on 13 or more pins.
Call `analogWriteResolution()` with different values.  Visually see by connecting LEDs to each of those 13 pins.

**Expected behavior**
All analog pins will update to reflect the specified resolution.

**Actual behavior**
The first 12 pins will update to reflect the specified resolution (COUNTERTOP changes), but the 13th pin will continue using the original PWM resolution.

hathach, 2020-03-20:
@henrygab yeah right, would you mind submitting PR to fix this ?

henrygab, 2020-03-20:
OK, PR for this single fix provided.

commit 8a1dff3f (2020-03-21): Merge pull request #468 from henrygab/FIX__458

Fix #458
commit c91bed14 (2020-03-21): Fix #458

Update HwPWM3 when it exists
closed 2020-03-21 by hathach