adafruit/Adafruit_nRF52_Arduino #460: bool Adafruit_LittleFS_Namespace::File::_open(const char*, uint8_t): 136: verify  failed, error = 2816

davPo, 2020-03-15:
**Describe the bug**
I can read / write with the LittleFS multiple times without any problem but after a BLE connection as peripheral, it is impossible to write using LittleFS again. I cannot identify the error code description.

bool Adafruit_LittleFS_Namespace::File::_open(const char*, uint8_t): 136: verify failed, error = 2816

**Set up (please complete the following information)**
  - NRF52832 Feather and BSP version 0.18.5
  - Windows 10
  - Debug Mode 2

My write function is inside the BLE callback but I have the same result if I call it manually through serial interface a long time after the BLE is disconnected.

Thanks for any support

hathach, 2020-03-19:
can you please post your simplest sketch and steps to reproduce the issue as well.

henrygab, 2020-03-24:
<HR/><details><summary>[Click to expand] The output indicates the source</summary><P>

```C++
bool Adafruit_LittleFS_Namespace::File::_open(const char*, uint8_t): 136: verify failed, error = 2816
```

That output is generated by the macro `PRINT_LFS_ERR()`, as used here:

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/110b06c8f0bd65aa668c2f48433144da4424afd8/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS_File.cpp#L121-L137

</details><HR/>

Therefore, the underlying error comes from line 122:
`int rc = lfs_stat(_fs->_getFS(), filepath, &info);`

Error codes generated by LittleFS are all negative values.  Presuming first that the LittleFS code is not at fault, the positive-valued error code would come from the Adafruit block layer.  Returning a positive error code would be a bug in the Adafruit block layer.  See LittleFS v2.1.0 [release notes](https://github.com/ARMmbed/littlefs/releases/tag/v2.1.0):

> Added asserts help stop positive error codes from block devices from causing the filesystem to break, this has been a reoccurring issue [#215](https://github.com/ARMmbed/littlefs/issues/215), [#169](https://github.com/ARMmbed/littlefs/issues/169)

Therefore, it might be fruitful to investigate the possiblity that Adafruit's block layer is returning a positive value as an error code.

davPo, 2020-03-25:
sorry for long delay, I finally found the issue in my code, I am copying received data to a tmp array in the callback function and the pointer was going to far (as often). Took me some time because my debugger could not break  systematically in this piece of code. I think I should close the issue ?

henrygab, 2020-03-25:
@davPo -- thanks for following up when you discovered it was in your own code.  This helps save precious time for other issues that are in the board.  I will close the issue.  Thanks again for following up so we can close this out!  👍

closed 2020-03-25 by henrygab