adafruit/Adafruit_nRF52_Arduino #462: Regex for arm-none-eabi-size output fails in arduino-cdt

toxynoid, 2020-03-19:
The regex for calculating the size with arm-none-eabi-size in platform.txt seem wrong to me
`recipe.size.regex=\.text\s+([0-9]+).*`
should be more
`recipe.size.regex=(?:\.text|\.data)\s+([0-9]+).*`
because as mentioned [here](https://mcuoneclipse.com/2013/04/14/text-data-and-bss-code-and-data-size-explained/) data is in the ROM as well. The non capturing regex group because the grouping match with index 1 gets evaluated in arduinocdt and there is only one capturing group in the regex for platform.txt in the arduino package as well.
This: 
`recipe.size.regex.data=(\.data|\.bss|\.stack)\s+([0-9]+).*`
should be:
`recipe.size.regex.data=(?:\.data|\.bss)\s+([0-9]+).*`
There is no line "stack" in the output of gcc size and again only second group should do capturing.

**Set up**
  - Adafruit nRF52 Boards Feather Express
  - 0.18.5
  - Ubuntu 19.10

hathach, 2020-03-20:
thanks for the issue, I am also long-time eclipse user, though I haven't used arduino cdt yet. Is it part of the cdt extension package. If you could give me brief on how to install it, I will try it out for confirmation

[image]

toxynoid, 2020-03-20:
Am 20.03.20 um 07:38 schrieb Ha Thach:
> thanks for the issue, I am also long-time eclipse user, though I haven't
> used arduino cdt yet. Is it part of the cdt extension package. If you
> could give me brief on how to install it, I will try it out for
confirmation

Just open
"help -> eclipse market place"
and install
"Eclipse C++ IDE for Arduino 3.0"
After restart you have
"help -> Arduino Downloads Manager"
from there you add the platform see screenshot arduino-dl-manager.png.
Then after you created a new arduino project right click project and
activate the libraries. See libs.png

The recipe.objcopy.zip.pattern from platform.txt does not get executed
by cdt yet. I'm planning on implementing it.

Right now zip task is planned as Makefile target but it seems that the
zip task would be better suited for the "tools" section of platform.txt.
So I'm not sure which to use.

Do you have further specifications how platform.txt is to be
edited/interpreted?

Best

hathach, 2020-03-20:
here is the specification for platorm.txt https://arduino.github.io/arduino-cli/platform-specification/

hathach, 2020-03-21:
it takes too long to download bsp, failed multiple times while downloading gcc tools. I guess there is a bug with arduino-cdt somehow. I will only review the changes for now, unfortunately.

hathach, 2020-03-24:
fixed by #472

closed 2020-03-24 by hathach