adafruit/Adafruit_nRF52_Arduino #475: nRF52840 Express Sketch Won't Run Without Serial Monitor

czuvich, 2020-04-02:
My sketches won't run on the nRF52840 Express unless the Serial monitor is open.

OS: MacOS Catalina
Arduino IDE: 1.8.10
BSP: 0.3.2

I tried uploading the Blinky sketch and the light does not blink unless I open the Serial monitor.  I downgraded the BSP to 0.3.0 and I don't see this issue.

jpconstantineau, 2020-04-02:
I have seen this on sketches where we have "while ( !Serial ) delay(10);".
Commenting it out resolved the issue.
Blinky doesn't have this line.
Has something changed in the bootloader to cause this?

hathach, 2020-04-03:
answered in the forum https://forums.adafruit.com/viewtopic.php?f=57&t=164090&p=806291#p806291

czuvich, 2020-04-06:
Not sure I really agree with this "feature", but ah well.  For me (and I'm probably not alone), I would like Serial output available in DEBUG mode even on a device disconnected from Serial.

The reason for that is if I encounter an issue while disconnected, it's nice to just connect to see Serial output.  That's very helpful without having to reflash the hardware while troubleshooting.  I'm not entirely sure why it was added (nor what the harm is in leaving it), but for now I guess I'll have to stick with the previous BSP.

dhalbert, 2020-04-06:
@czuvich Are you setting the Arduino Tools->Debug menu level to something other than 0? If it's 0, then it should not hang.

@hathach Suppose there was a timeout on the `while ( !Serial ) yield();` so that the program would proceed after a few seconds? Then running a program on battery power with Debug set would still work, with a startup delay.

I agree that raising the debug level as you mentioned in the forum post may also be a good idea.

czuvich, 2020-04-06:
> @czuvich Are you setting the Arduino Tools->Debug menu level to something other than 0? If it's 0, then it should not hang.

See my above post.  I now understand the issue is setting the DEBUG level and Serial; however, I personally think it makes troubleshooting more cumbersome.  It'd be nice to have Serial output available without being connected to Serial (i.e. battery power) while still having full DEBUG output.  LOG_LV1/LV2 don't really output much on the bootloader, so why not let the developer leverage it.

> @hathach Suppose there was a timeout on the while ( !Serial ) yield(); so that the program would proceed after a few seconds? 

A happy compromise.  I still find it strange though why this feature was added in the first place.  Again.. thanks for the quick replies and help.

henrygab, 2020-04-12:
@hathach - two thoughts that both suggest removing this is the wiser course:

1. Can you help me understand the benefit of having the BSP do this, vs. a sketch doing the same `while (!Serial) yield();` itself?
  I ask because I have not yet understood the benefit of forcing this behavior in the BSP.  Does this provide an advantage or prevent some weird edge case where the user is otherwise unable to get their setup() routine to run?

2. I also was bitten by this behavior recently.  However, I didn't know what was causing it, and lost quite a few hours (effectively days in real time), because I simply did not expect the board to lock up without a serial connection.  I full-erased the board more than once, reloaded bootloaders, ... and then stumbled on this thread, showing the true cause.  This was frustrating, and there is no real way to inform the user to help them troubleshoot if they hit this problem.

At present, I'm strongly in favor of removing this, and allowing any sketch that wants to wait for serial to perform this loop itself.  However, I recognize there may be other aspects that require this, so seek your knowledge.

Of course, if this isn't required, please PLEASE consider removal.  It is unbelievably frustrating to get bitten by this problem.

Thank you!

hathach, 2020-04-13:
> @hathach - two thoughts that both suggest removing this is the wiser course:
>
> 1. Can you help me understand the benefit of having the BSP do this, vs. a sketch doing the same `while (!Serial) [...]

Ok, Ok I will remove it, it turns out not a good thing to have :)

closed 2020-04-13 by hathach

czuvich, 2020-04-13:
@henrygab I need to learn your communication skills lol.  Thanks for the fix.

hathach, 2020-04-13:
> @henrygab I need to learn your communication skills lol. Thanks for the fix.

He is an excellent writer, unfortunately as non-English speaker, I feel so overwhelmed by that amount of English words. Just kidding :laughing: :laughing:

PR #481 (2020-04-13): remove the force Serial connected when debug is eanbled.
fix #475 , per user feedback. Move the blocking to user sketch.