adafruit/Adafruit_nRF52_Arduino #476: Unable to upload any sketch to ItsyBitsy nRF52840 using `adafruit-nrfutil` in `arduino-cli`

ardnew, 2020-04-04:
**Describe the bug**
When trying to upload any sketch using the official [arduino-cli](https://github.com/arduino/arduino-cli), the `adafruit-nrfutil` upload utility fails to locate the generated DFU .zip package, and thus can't program the device.

**Set up (please complete the following information)**
  - ItsyBitsy nRF52840, `adafruit:nrf52 0.19.0`, `arduino-cli Version: 0.0.0-git Commit: 8759bf46`
  - `Ubuntu 18.04.4 LTS`
```
== SETTINGS ======================================================

  config         - --config-file /home/andrew/.config/arduino-cli/config.yaml
  mode           - compile+upload
  board (fqbn)   - adafruit:nrf52:itsybitsy52840
  upload         - /dev/ttyACM1
  sketchbook     - /home/andrew/Development/arduino/sketchbook
  sketch         - bled
  log level      - trace

==================================================================

  arduino-cli --config-file /home/andrew/.config/arduino-cli/config.yaml compile
    --verbose --log-level trace
    --fqbn adafruit:nrf52:itsybitsy52840
    --upload --port /dev/ttyACM1
    --build-path /home/andrew/.tmp/arduino-build/bled
    --build-cache-path /home/andrew/.tmp/arduino-build.cache/bled
      /home/andrew/Development/arduino/sketchbook/bled

==================================================================

------------------------------------------------------------------
$ arduino-cli --config-file /home/andrew/.config/arduino-cli/config.yaml compile --verbose --log-level trace --fqbn adafruit:nrf52:itsybitsy52840 --upload --port /dev/ttyACM1  --build-path /home/andrew/.tmp/arduino-build/bled --build-cache-path /home/andrew/.tmp/arduino-build.cache/bled  /home/andrew/Development/arduino/sketchbook/bled
------------------------------------------------------------------

Using board 'itsybitsy52840' from platform in folder: /home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0
Using core 'nRF5' from platform in folder: /home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0
Detecting libraries used...
/home/andrew/.config/arduino-cli/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=64000000 -DARDUINO=10607 -DARDUINO_NRF52840_ITSYBITSY -DARDUINO_ARCH_NRF52 "-DARDUINO_BSP_VERSION=\"0.19.0\"" -DNRF52840_XXAA -DARDUINO_NRF52_ITSYBITSY -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8051 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"ItsyBitsy nRF52840 Express\"" -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/cmsis/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/hal -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/mdk -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/soc -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/drivers/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/drivers/src -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/Source/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/config -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/portable/GCC/nrf52 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/portable/CMSIS/nrf52 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/sysview/SEGGER -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/sysview/Config -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/TinyUSB -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/variants/itsybitsy_nrf52840_express /home/andrew/.tmp/arduino-build/bled/sketch/bled.ino.cpp -o /dev/null
Alternatives for Adafruit_NeoPixel.h: [Adafruit_NeoPixel@1.3.5]
ResolveLibrary(Adafruit_NeoPixel.h)
  -> candidates: [Adafruit_NeoPixel@1.3.5]
/home/andrew/.config/arduino-cli/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=64000000 -DARDUINO=10607 -DARDUINO_NRF52840_ITSYBITSY -DARDUINO_ARCH_NRF52 "-DARDUINO_BSP_VERSION=\"0.19.0\"" -DNRF52840_XXAA -DARDUINO_NRF52_ITSYBITSY -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8051 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"ItsyBitsy nRF52840 Express\"" -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/cmsis/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/hal -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/mdk -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/soc -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/drivers/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/nrfx/drivers/src -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/Source/include -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/config -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/portable/GCC/nrf52 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/freertos/portable/CMSIS/nrf52 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/sysview/SEGGER -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/sysview/Config -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/TinyUSB -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/cores/nRF5 -I/home/andrew/.config/arduino-cli/packages/adafruit/hardware/nrf52/0.19.0/variants/itsybitsy_nrf52840_express -I/home/andrew/Development/arduino/sketchbook/libraries/Adafruit_NeoPixel /home/andrew/.tmp/arduino-build/bled/sketch/bled.ino.cpp -o /dev/null
Using cached library dependencies for file: /home/andrew/Development/arduino/sketchbook/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp
Using cached library dependencies for file: /home/andrew/Development/arduino/sketchbook/libraries/Adafruit_NeoPixel/esp8266.c
[... cut ...]
N/A

**Additional context**
This issue was also posted at https://github.com/arduino/arduino-cli/issues/641 because I'm not really sure which project is responsible. There is a little more detail in diagnosing the issue in that thread.

If I manually upload using the utility template and provide the DFU .zip path explicitly, the programming works as expected:
```
adafruit-nrfutil --verbose dfu serial -pkg /home/andrew/.tmp/arduino-build/bled/bled.ino.zip -p /dev/ttyACM1 -b 115200 --singlebank
```

hathach, 2020-08-03:
I have tried with arduino-cli 0.11.0 , seems like they fix the issue, it is worth to give it a try again. 

```
arduino-cli Version: 0.11.0 Commit: 0296f4d
```

Tested with much simpler command that take the sketch folder as parameter. Note: if you have issue with arduino-cli cannot perform DTR, please update to master which add a few ms delay before reset into DFU #549

```
arduino-cli compile --fqbn adafruit:nrf52:feather52840:softdevice=s140v6,debug=l0 --upload --port /dev/ttyACM0 sketch_jul29a

Sketch uses 74116 bytes (9%) of program storage space. Maximum is 815104 bytes.
Global variables use 11996 bytes (5%) of dynamic memory, leaving 225572 bytes for local variables. Maximum is 237568 bytes.
Upgrading target on /dev/ttyACM0 with DFU package /home/hathach/Arduino/sketches/sketch_jul29a/build/adafruit.nrf52.feather52840/sketch_jul29a.ino.zip. Flow control is disabled, Single bank, Touch disabled
########################################
########################################
########################################
#########################
Activating new firmware
Device programmed.
```

hathach, 2020-08-03:
closed since not reproducible.

closed 2020-08-03 by hathach