adafruit/Adafruit_nRF52_Arduino #477: GPIOTE channels are overwritten if in use by library

henrygab, 2020-04-11:
**Describe the bug**
If a library uses a GPIOTE channel 0, and then the sketch calls attachInterrupt(), the library's configuration is corrupted.

**Set up (please complete the following information)**
  - nRF52840 Feather Express
  - BSP from commit 0019306c (0.19.0)
  - Windows 10
  - Log: N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Configure `GPIOTE` channel 0 and `PPI` channels 0-4 (e.g., as a user, using an opaque library) to configure GPIO output state changes without CPU involvement
2. Once that sketch is working, add button on another GPIO, using `attachInterrupt()` to register an interrupt handler for that button
3. Sadness, as things stop working
4. Comment out call to `attachInterrupt()`
5. Happiness, as sketch starts working again!
6. Track down underlying issue, and realize the cause is similar to #459 and #447

**Expected behavior**
BSP doesn't use peripherals that are clearly already in use.

henrygab, 2020-04-12:
It appears the following would also be problematic:
```C++
// sketch initially looks for a rising edge on a first pin (alpha)
attachInterrupt(PIN_ALPHA, alpha_isr, RISING);
// and also looks for a rising edge on a second pin (omega)
attachInterrupt(PIN_OMEGA, omega_isr_rise, RISING);
// ... eventually, sketch no longer needs interrupts on pin alpha
detachInterrupt(PIN_ALPHA);
// ... and later still, sketch modifies omega to interrupt on both the rising and falling edges ...
attachInterrupt(PIN_OMEGA, omega_isr_change, CHANGE);
// ... finally, still later, the sketch no longer needs interrupts on pin alpha
detachInterrupt(PIN_OMEGA);
```

<HR/><details><summary>[click to expand] Problem walkthrough</summary><P/>

What is the problem?  Starting from a fresh system:
```C++
// 0. Fresh system -- no GPIOTE Channels in use

// 1. GPIOTE channel 0 is assigned to PIN_ALPHA
attachInterrupt(PIN_ALPHA, alpha_isr, RISING);

// 2. GPIOTE channel 1 is assigned to PIN_OMEGA (thus ch 0 & 1 in use)
attachInterrupt(PIN_OMEGA, omega_isr_rise, RISING);

// 3. GPIOTE channel 0 is released (thus ch 1 remains assigned to OMEGA)
detachInterrupt(PIN_ALPHA);

// 4. GPIOTE channel 0 gets assigned to OMEGA, as the first GPIOTE that is not currently use, thus:
//     ch 0 is assigned to OMEGA (RISING)
//     ch 1 is assigned to OMEGA (CHANGE)
attachInterrupt(PIN_OMEGA, omega_isr_fall, CHANGE);

// 5. GPIOTE channel 0 is released
//     but, ch 1 is still assigned to OMEGA (CHANGE)
detachInterrupt(PIN_OMEGA);
```
Step 4 appears to result in undefined behavior, because two GPIOTE channels are both configured against a single GPIO pin.

Step 5 also results in "bad stuff".  At best, a GPIOTE channel is leaked and cannot be used by another pin.
</details

Anyways, I'll fix that too, by first looping through all GPIOTE channels to see if the pin is already assigned to one of the GPIOTE channels.  Only when the pin is not already assigned a GPIOTE, a second loop runs to attempt to allocate an unused GPIOTE channel.

hathach, 2020-04-14:
@henrygab thank you very much for your effort explaining thing very detail. Unfortunately, I am not really good at reading. I am still confused and not really sure where is the problem with GPIOTE. Is that attachInterrupt() with same pin in different configure, if so, I will say it is illegal. Le'ts sketch dettach it first.

PS: I would prefer an runnable sketch in case I want to test it out.

hathach, 2020-04-15:
Sorry, my head was spinning spending too much time debugging. This apparently needs further reading from my side. To be honest, I am not familiar with  GPIOTE at all, this file is from original repo. Though I will try to catch up with up later on. Thank you very much for filing the issue :)

henrygab, 2020-04-15:
Hi @hathach,

OK, I have created a minimal bug repro sketch for you:
https://gist.github.com/henrygab/d02e6b03be5636d51fc0fe8748cb3b21

<details><summary>[click to expand] Using the repro sketch</summary><P/>

You may want to set break-on-write breakpoints on `channelMap[0]` and `channelMap[1]`, or at least set them as watch variables.

1. Jumper the two pairs of pins as noted at the top of the file
2. Compile and upload to Adafruit nRF52840 Feather Express
3. See how things should look when code acts as it should
    * This is the behavior in most situations, and is the target behavior for the fix
4. Modify the variable `causeBug` to have the value `1`, re-compile, and upload
    * Note: this only changes the order of which pin is sent to `attachInterrupt()` first
5. See how badly things go wrong....

I tried to show in the comments how the `channelMap` array in `attachInterrupt()` looks through the sketch.

Existing code behavior generally is as follows:
* a call to `attachInterrupt()` for a pin that's already attached gets the update mode and callback applied to it
* a call to `detachInterrupt()`  removes the corresponding entry from `channelMap`

</details><HR/>
<details><summary>[click to expand] How those two bugs are fixed</summary><P/>

Loop through the `channelMap` array **_twice_**:

1. the first time, search for the pin already being in **_ANY_** index of `channelMap`, even if unused entries exist earlier.
2. the second time, search only for an unused entry in the `channelMap`, to indicate adding a new pin.

In this way, it's ensured that a single pin will not be listed twice in the `channelMap` array.
</details><HR/>

<details><summary>[click to expand] Another bug fixed with this PR</summary><P/>

Previously, `attachInterrupt()` would also stomp on any other code's use of a GPIOTE channel.
This PR will include the check for (and skip) GPIOTE channels that are already in use, regardless of what code configured the GPIOTE channel for use.

See the second loop.

</details><HR/>

NOTE: This is a WIP PR ... I am still validating behavior and fix for the second bug, although it's already included in the WIP PR, so you can review the concept now.

henrygab, 2020-04-15:
@hathach -- no problem.  the example sketch should make some it clear.

GPIOTE has three purposes:
1. Use to enable interrupts, as is being used in the code being fixed
2. Use to configure GPIO input pin as trigger event (on rise, fall, or change), such as in connection with a PPI channel (e.g., zero-CPU triggering of peripheral events) 
3. Use to configure GPIO output pin, so PPI channel can control the GPIO output (e.g., zero-CPU toggle of GPIO pin based on peripheral events)

hathach, 2020-04-15:
> @hathach -- no problem. the example sketch should make some it clear.
>
> GPIOTE has three purposes:
>
> 1. Use to enable interrupts, as is being used in the code being fixed
> 2. Use to configure GPIO [...]

Thank you, I will check it out (soooonn :) )

hathach, 2020-04-21:
> @hathach -- no problem. the example sketch should make some it clear.
>
> GPIOTE has three purposes:
>
> 1. Use to enable interrupts, as is being used in the code being fixed
> 2. Use to configure GPIO [...]

Sorry @henrygab I was drown with TinyUSB work. I have tried your excellent example sketch with very detailed comment for what it does and what is the expected result. I have a question though, would the issue be prevented if `PIN_OMEGA` is dettached before changing its events. 
https://gist.github.com/henrygab/d02e6b03be5636d51fc0fe8748cb3b21#file-bug_480-ino-L98 

If it is the case, then user should actually dettach() before changing its behavior. Arduino sketches are often very specific, and arduino library isn't written as bullet-proof. I am ok if imposing limit on such as flow chart on the sketch.

henrygab, 2020-04-21:
@hathach -- I can imagine the work ... bringing up new hardware is always a big task, and often made difficult by discovery of hardware errata.  I hope you had success.

If your question is whether this could be treated similar to a hardware errata, by having a release note or similar: Yes ... but not really.

If your question is whether this could be enforced in the API: No, I do not think that is a fix.

<HR/>
<details><summary>Driving factors:</summary><P>

1. Arduino is a learning platform.  This means it's critical that people can reason about what will happen (and what has happened) based on the API abstraction (i.e., without diving into BSP code).
2. If a sketch is working, ensuring backwards compatibility is a high priority.  When necessary to break the backwards compatibility, at least one of the following are strongly desired, in order of preference:
    a. Compile-time break, such as via a `static_assert()` that points to actions to be taken to fix
    b. Run-time debug message and assertion, that points to actions to be taken to fix
3. Arduino is meant to enable projects on one board to transition to another board, with minimal change when using the Arduino APIs.

</details><HR/>
<details><summary>Why not really</summary><P>

1. The `attachInterrupt()` code has special-case for exactly the case you suggest should be disallowed.  Therefore, it's likely an **_intentional feature_**, and not a bug, to allow changing from edge to falling (or vice versa), or to allow changing from edge to rising (or vice versa).

2. The `attachInterrupt()` API is designed to mirror registration of a hardware interrupt; That is, one (and only one) ISR is intended to be allowed per pin.

3. Every bug could be worked around by simply saying "on this board, don't do this sequence of things, because it causes bug XYZ".  Similar to hardware errata, this makes using the BSP less pleasant.  Unlike hardware errata, where there is no other option, this is not really a preferred solution for software. 

4. Here, as seen in the sketch, something apparently unrelated (which pin's interrupt is enabled first) causes things to break.  This prevents reasoning about the system's behavior, without diving deep into the BSP specific implementation.  Having a note somewhere indicating this bug exists does not 

5. Imposing a new limit to prevent changing from edge to rising can break existing, working sketches.  It also appears to prevent a seamless transition (risk of lost interrupts) between edge detection modes.

</details><HR/>
<details><summary>The bug isn't that complex...</summary><P/>

The bug is actually quite simple.  Just because most times, the array is packed, the code incorrectly makes the assumption that, if it finds any element not in use, it presumes that there are no further elements in use later in the array.

Of course, the array is not guaranteed to be packed.  It's quite easy to cause the first element to be empty (`-1`) while the second element has a valid value.  For example, the following is clearly 100% legal, yet causes this situation:
```C++
attachInterrupt(PinA, ...);
attachInterrupt(PinB, ...);
detachInterrupt(PinA);
```

Similarly, although I did comment that the trouble started at [line 98](https://gist.github.com/henrygab/d02e6b03be5636d51fc0fe8748cb3b21#file-bug_480-ino-L98), that was just the trouble **_that the user experienced_**.  The bug can manifest anytime after [lines 75-76](https://gist.github.com/henrygab/d02e6b03be5636d51fc0fe8748cb3b21#file-bug_480-ino-L75-L76).

</details><HR/>

I believe the concept shown in my branch will provide this.  But, I've not time to test and verify the fix yet ... I'm sure there are still bugs.

hathach, 2020-04-22:
@henrygab OK, that is too many English text, I will check out your PR instead, since I am more comfortable reading C code. just kidding :smile:

PR #485 (2020-04-28): GPIOTE fixes for attachInterrupt() / detachInterrupt()
# WORK IN PROGRESS [WIP]
Fix  #477

Although the bug is confirmed (see [gist](https://gist.github.com/henrygab/d02e6b03be5636d51fc0fe8748cb3b21)), the fix is not fully ready.

Although the concept is clear from the first two commits in this branch, there is still at least one bug that prevents integration at the moment.

tagging myself @henrygab
closed 2020-04-28 by hathach