adafruit/Adafruit_nRF52_Arduino #478: Indicate property does not seem to be working

jeffdeville, 2020-04-11:
**Describe the bug**
the custom_htm demo on my nRF52840 reports: "Error: Indicate not set in the CCCD or not connected!"

custom_hrm (which uses notify) works fine though.  

I'm testing with BLE Scanner 4.0 for ios.

**Set up (please complete the following information)**
  - Your Board and BSP version
nRF5240, I've pulled down the libraries directly from github on master. 
0.19.0 is the version of Adafruit nRF52 reported in my Boards Manager

  - Your OS (mac/windows/linux) and its version
OSX - Catalina 10.15.3

  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  

10:22:55.887 -> Advertising!
10:22:58.882 -> Connected to iPhone
10:22:59.881 -> ERROR: Indicate not set in the CCCD or not connected!

**To Reproduce**
Steps to reproduce the behavior:

1. load custom_htm into the Arduino IDE
2. Deploy it to a nRF52840
3. Load the serial monitor
4. Connect with BLE Scanner (iOS)

**Expected behavior**

I expect the Health Thermometer service to have a property for the temperature that is set, and updates. But no value is ever set. And I think the reason why is related to the fact that the CccdWriteCallback is never invoked for INDICATE properties, but is for NOTIFY properties.

**Additional context**

Happy to provide any more detail that would help. I'm new to C++/Arduino, but have been programming for quite some time, so I can hopefully muddle through and figure out how to get you whatever else would help.

hathach, 2020-04-14:
please try the nrftoolbox app, there is a htm module there. Let's me know if that works out. if it does then the problem is within your current client app.
https://apps.apple.com/us/app/nrf-toolbox/id820906058

jeffdeville, 2020-04-17:
Oh wow, you're quite right @hathach! Thank you!

closed 2020-04-17 by jeffdeville