adafruit/Adafruit_nRF52_Arduino #483: configUuid128Count() causes failure

shtarbanov, 2020-04-13:
Setting the argument of configUuid128Count() to anything greater than the default value of 10 causes bluetooth to fail. Even a value of 11 does not work.

shtarbanov, 2020-04-13:
I have looked at the definition of the function

void AdafruitBluefruit::configUuid128Count(uint8_t  uuid128_max)
{
  _sd_cfg.uuid128_max = uuid128_max;
}

 But I cannot find where is "_sd_cfg.uuid128_max" defined? 
Where can I find any information about "_sd_cfg" itself?

hathach, 2020-04-14:
please help us to provide enough info to reproduce and troubleshoot 
- fill out the requirement information for the hw setup 
- provide your simplest sketch to reproduce
- provide your serial output in debug mode.

shtarbanov, 2020-04-16:
Upon further investigation, I found out that the problem occurs when only when BOTH of these functions are used in the same sketch:

Bluefruit.configAttrTableSize();  //I set the argument of this to 0xD60
Bluefruit.configUuid128Count();

To replicate the problem, open any of the included peripheral examples and add the following lines of code. For instance, you can open the bleuart.ino example and right before the Bluefrut.begin() add the lines:

  Bluefruit.configAttrTableSize(0xD60);
  Bluefruit.configUuid128Count(11);

Everything will compile ok and upload fine to the board. But after upload, nothing is being broadcasted by the device. If you change the value to 10, or if you remove the first function, then it all works again.

hathach, 2020-04-16:
Please provide all the information I asked in the above comment. The sketch should be as minimal as it could get.

hathach, 2020-07-14:
closed due to lack of activities

closed 2020-07-14 by hathach