adafruit/Adafruit_nRF52_Arduino #488: Cannot get example sketches to compile on nRF52840 feather

cyborg5, 2020-04-21:
I've tried to get multiple examples sketches to run on my Feather nRF52840 Express and none of them can compile. Specifically I tried to compile
[central_hid](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central/central_hid)

[hid_keyboard](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard)

[beacon](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Peripheral/beacon)

Here are the error messages for beacon

```
Arduino: 1.8.12 (Windows 10), Board: "Adafruit Feather nRF52840 Express, 0.3.2 SoftDevice s140 6.1.1, Level 0 (Release)"

In file included from C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid_device.h:32:0,

                 from C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/tusb.h:68,

                 from C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/Adafruit_TinyUSB_Core.h:32,

                 from C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/Arduino.h:71,

                 from C:\Users\Chris.cy\Documents\Arduino\libraries\Bluefruit52Lib\src/bluefruit.h:27,

                 from C:\Users\Chris.cy\Documents\Arduino\libraries\Bluefruit52Lib\src\services\BLEHidAdafruit.cpp:37:

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid.h:339:41: error: 'REPORT_ID_KEYBOARD' cannot be used as a function

   (((tag) << 4) | ((type) << 2) | (size)) HID_REPORT_DATA_##size(data)

                                         ^

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid.h:392:35: note: in expansion of macro 'HID_REPORT_ITEM'

 #define HID_USAGE_PAGE(x)         HID_REPORT_ITEM(x, 0, RI_TYPE_GLOBAL, 1)

                                   ^~~~~~~~~~~~~~~

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid_device.h:118:5: note: in expansion of macro 'HID_USAGE_PAGE'

     HID_USAGE_PAGE ( HID_USAGE_PAGE_KEYBOARD )                     ,\

     ^~~~~~~~~~~~~~

C:\Users\Chris.cy\Documents\Arduino\libraries\Bluefruit52Lib\src\services\BLEHidAdafruit.cpp:52:3: note: in expansion of macro 'TUD_HID_REPORT_DESC_KEYBOARD'

   TUD_HID_REPORT_DESC_KEYBOARD( HID_REPORT_ID(REPORT_ID_KEYBOARD) ),

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid.h:339:41: error: 'REPORT_ID_CONSUMER_CONTROL' cannot be used as a function

   (((tag) << 4) | ((type) << 2) | (size)) HID_REPORT_DATA_##size(data)

                                         ^

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid.h:395:35: note: in expansion of macro 'HID_REPORT_ITEM'

 #define HID_LOGICAL_MIN(x)        HID_REPORT_ITEM(x, 1, RI_TYPE_GLOBAL, 1)

                                   ^~~~~~~~~~~~~~~

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid_device.h:208:5: note: in expansion of macro 'HID_LOGICAL_MIN'

     HID_LOGICAL_MIN  ( 0x00                                ) ,\

     ^~~~~~~~~~~~~~~

C:\Users\Chris.cy\Documents\Arduino\libraries\Bluefruit52Lib\src\services\BLEHidAdafruit.cpp:53:3: note: in expansion of macro 'TUD_HID_REPORT_DESC_CONSUMER'

   TUD_HID_REPORT_DESC_CONSUMER( HID_REPORT_ID(REPORT_ID_CONSUMER_CONTROL) ),

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid.h:339:41: error: 'REPORT_ID_MOUSE' cannot be used as a function

   (((tag) << 4) | ((type) << 2) | (size)) HID_REPORT_DATA_##size(data)

                                         ^

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid.h:426:35: note: in expansion of macro 'HID_REPORT_ITEM'

 #define HID_USAGE(x)              HID_REPORT_ITEM(x, 0, RI_TYPE_LOCAL, 1)

                                   ^~~~~~~~~~~~~~~

C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\cores\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/hid/hid_device.h:159:5: note: in expansion of macro 'HID_USAGE'

     HID_USAGE      ( HID_USAGE_DESKTOP_POINTER )                   ,\

     ^~~~~~~~~

C:\Users\Chris.cy\Documents\Arduino\libraries\Bluefruit52Lib\src\services\BLEHidAdafruit.cpp:54:3: note: in expansion of macro 'TUD_HID_REPORT_DESC_MOUSE'

   TUD_HID_REPORT_DESC_MOUSE   ( HID_REPORT_ID(REPORT_ID_MOUSE) )

   ^~~~~~~~~~~~~~~~~~~~~~~~~

Multiple libraries were found for "bluefruit.h"
 Used: C:\Users\Chris.cy\Documents\Arduino\libraries\Bluefruit52Lib
 Not used: C:\Users\Chris.cy\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.19.0\libraries\Bluefruit52Lib
exit status 1
Error compiling for board Adafruit Feather nRF52840 Express.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

hathach, 2020-04-22:
@cyborg5 hi hi sorry for the issue, are you using the latest BSP nRF52 from master. I have just updated the tinyusb stack to the latest. Which contains an **FIX** for hid report descriptor, it make the hid report descriptor easier on eye. Please notice the extra comma after the HID_REPORT_ID() is not needed with the update.
```
uint8_t const desc_hid_report[] =
{
  TUD_HID_REPORT_DESC_KEYBOARD( HID_REPORT_ID(RID_KEYBOARD),  ),
  TUD_HID_REPORT_DESC_MOUSE   ( HID_REPORT_ID(RID_MOUSE),  ),
};

to

uint8_t const desc_hid_report[] =
{
  TUD_HID_REPORT_DESC_KEYBOARD( HID_REPORT_ID(RID_KEYBOARD) ),
  TUD_HID_REPORT_DESC_MOUSE   ( HID_REPORT_ID(RID_MOUSE) ),
};

```
There is a PR to correct this compile issue https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/56/files . But it needs to wait until the BSP for both SAMD and nRF to roll out. Which I am about to do today. Meanwhile you can simply remove the extra **comma** in the hid descriptor and it should be compile again. Sorry for the inconvenience.

hathach, 2020-04-23:
@cyborg5 I have release both BSP nRF52840 0.20.0 and TinyUSB library 0.9.0 . Could you please update and try it again to see if that fixes your issue.

hathach, 2020-05-08:
should be fixed by now.

closed 2020-05-08 by hathach