adafruit/Adafruit_nRF52_Arduino #490: Arduino-cli fails to install adafruit:nrf52@0.20.0 with "Error during install: extracting archive: Not a supported archive

jpconstantineau, 2020-04-23:
**Describe the bug**
A clear and concise description of what the bug is.

**Set up (please complete the following information)**
  - Your Board and BSP version
  - Your OS (mac/windows/linux) and its version
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jpconstantineau, 2020-04-23:
Sorry about that.  for some reason the issue when in before typing anything.  I re-ran and it installed fine.

hathach, 2020-04-23:
no problems, it is just released, there may be a bit of issue.

jpconstantineau, 2020-04-23:
I have a docker image that wraps the arduino-cli and the nrf52 BSP so that
I can test my firmware.  Last time it ran, it failed with this obscure
archive file error.  re-running the docker build to build a new image ran
fine...

I don't see the BSP in the github releases yet (
https://github.com/adafruit/Adafruit_nRF52_Arduino/releases).  Is that
something that you do manually? or done through your builds?

hathach, 2020-04-23:
we use action ci verify before and after releasing. We will know if there is something wrong with the release.