adafruit/Adafruit_nRF52_Arduino #492: No Serial Output on Beacon example

rwarrender, 2020-04-24:
**Describe the bug**
No Serial output and beacon not active

**Set up (please complete the following information)**
  - Adafruit Feature Express nRF52840 Express
  - Mac OS X 10.15.3
  - Arduino 1.8.12  
  - Debug log set to 2

***Bootloader Info***
```UF2 Bootloader 0.3.2 lib/nrfx (v2.0.0) lib/tinyusb (legacy-1500-g23df777b) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: nRF52840-Feather-revD
Date: Mar 12 2020```

**To Reproduce**
Steps to reproduce the behavior:
1. Open Arduino
2. File > Examples > Adafruit Bluefruit nF52 Libraries > Peripheral >  beacon
3. Click Upload Button
4. Open console log once successfully uploaded

**Expected behavior**
Console log messages printed and beacon discoverable.

**Additional context**
Add any other context about the problem here.

hathach, 2020-04-24:
could you tell me your BSP version ?

rwarrender, 2020-04-24:
Hi sorry, it's Adafruit nRF52 0.20.1

hathach, 2020-04-25:
@rwarrender no problem, I just tested the sketch, it runs fine. The output doesn't got printed since the loop got suspended by the end of setup(). The beacon is advertising, you could see if using nrfconnect or an scanner, or compatible beacon app. I used to use nrf beacon app, but forgot the exact one. If you add an small delay before suspendLoop(); or Serial.flush(), it should come out. I will try to improve the usb serial, but that is for another issue.

```
delay(1); // or Serial.flush()
suspendLoop();
```

[image]

rwarrender, 2020-04-25:
@hathach thanks, I get output in the console now and the beacon does work now! I thought I was going crazy!!

However I get this in the console: 

```Bluefruit52 Beacon Example
--------------------------

[CFG   ] SoftDevice's RAM requires: 0x20002D30
Broadcasting beacon, open your beacon app to test
```

Is `[CFG   ] SoftDevice's RAM requires: 0x20002D30` an error?

rwarrender, 2020-04-25:
Also I noticed that if we go back to using the original example sketch without the delay, that even though the console output won't display, the beacon won't run unless I open up the console

hathach, 2020-04-25:
because the sketch is waiting for the connection from Serial 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/beacon/beacon.ino#L37

I will slow remove those in example sketch. If the issue is fixed, please close it.

rwarrender, 2020-04-25:
Thanks for clarifying and investigating

closed 2020-04-25 by rwarrender