adafruit/Adafruit_nRF52_Arduino #495: External Voltage Reference Support for nRF52840

Montvydas, 2020-05-03:
**The Problem**
I've been working on Xenon based project, which requires knowing accurately measured voltage. I've read forums and found that nRF52840 internal voltage reference has poor tolerance, thus I intended using external voltage reference. After looking into the [schematics](https://github.com/particle-iot/xenon/blob/master/pdfs/xenon-v1.00-schematic.pdf) I found that pin AREF (P0.02) on Xenon is actually not exposed for us to be used...

I started checking on how this is done on Feather nRF52840 Express as it does have Aref pin and it's use explained in [pinout page](https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/pinouts) but didn't notice any example code. I also noticed that there wasn't any parameter which I could pass into the function analogReference(TYPE), which would start using Aref as a reference pin. TYPE here can only be AR_INTERNAL, AR_INTERNAL_3_0, AR_INTERNAL_2_4, AR_INTERNAL_1_8, AR_INTERNAL_1_2 and AR_VDD4. After checking the [nRF52845 docs](https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.1.pdf#page=126) I noticed the registers EXTREFSEL and REFSEL, which can be manipulated to use any of the analog pins to work as external voltage refs (that's great!). But when I checked the Arduino source code I noticed that the implementation is somewhat there but unfinished.. Check value _COMP_REFSEL_REFSEL_ARef_ == 5 in [nrf52840_bitfields.h](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/nordic/nrfx/mdk/nrf52840_bitfields.h), which coincides with the docs. This is later assigned to enum _nrf_comp_ref_t_ variable  _NRF_COMP_REF_ARef_  in file [nrf_comp.h](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/nordic/nrfx/hal/nrf_comp.h). In the same file noticed struct _nrf_comp_ext_ref_t_, which allows selecting which of the analog pins to use as an external ref. In the same file function _nrf_comp_ref_set(...)_ allows passing struct _nrf_comp_ref_t_ to set the reference type and function _nrf_comp_ext_ref_set(...)_ allows specifying which analog pin to use as a reference (the other function parameter I see is _NRF_COMP_Type_, which struct is defined in [nrf52840.h](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/nordic/nrfx/mdk/nrf52840.h) and I see that it has values EXTREFSEL and REFSEL, I think we're close!). There is also a struct _nrf_comp_ref_conf_t_, which holds both the reference type and which analog pin to use.

Ok now to the point, after greping all of the source code I didn't see any place where the two functions would be called from nor the struct _nrf_comp_ref_conf_t_ used. Therefore if I am not mistaken I see the initial code is there but it's incomplete even for using the Aref on Feather Express.

**The Solution**
I believe this could be solved in a couple of ways, but the preferred one is adding one more __eAnalogReference_ value to the struct called _EXTERNAL_, which could be then set through _analogReference(EXTERNAL)_ function. By default External reference would be used on analog pin A7 (aka Aref) unless modified before, in that case use the previously defined analog pin as a reference. Another function could be created named _externalReferencePin(_eExternalReferencePin)_, where you could pass a struct holding values A0-A7 and AREF/DEFAULT. This would be self documenting enough to be used and would coincide with the existing implementation on Arduino based on ATmega boards where EXTERNAL can be used as one of the options for analog reference.

**Workarounds?**
I understand this would take some time to implement. Hell, if it's nothing horribly complex I can implement it myself! Anyway for the moment being do you think there are any workarounds for this e.g. calling _nrf_comp_ref_set_ and _nrf_comp_ext_ref_set_ functions from the Arduino sketch itself?

Montvydas, 2020-05-03:
Ok whoops, I started going through all the code and docs and realised that external reference is only applicable for comparator mode... So basically nRF52840 Does not support External Analog Reference - I wonder why Feather nRF52840 Express pinout page is telling these horrible lies and even naming the A7 pin as Aref...??
What do you think is the best way to combat the not accurate internal reference? I was thinking of either using an external ADC or powering the chip from an external voltage reference and then using Vdd as a voltage reference...

dhalbert, 2020-05-03:
I don't think we are telling "horrible lies", but I'll add an explanation to the pinouts information that AREF is only for the COMP peripheral, not SAADC.

I helped someone else recently who was trying to do accurate measurements with the nRF52840 ADC and seeing significant jitter and some non-linearity. This was without oversampling. If you want more accurate results, then using an external ADC sounds like a good idea. Is it accuracy or stability you're looking for? The VDD reference could be helpful, but it depends on the regulator stability.

hathach, 2021-09-13:
Closed now since It is more related to hardware than software. You could open an issue on Adafruit support forum or Nordic one where you think is appropriate for further discussion

closed 2021-09-13 by hathach