adafruit/Adafruit_nRF52_Arduino #497: [Feature] Migrate Interrupt handling to nrfx

czuvich, 2020-05-05:
I was trying to use the nrfx_gpiote functionality, and I was getting a compiler error saying that "GPIOTE_IRQHandler was already defined".  I dug into WInterrupts.c, and I noticed it is using the legacy drivers to configure interrupts.  nrfx_gpiote.h also defines the GPIOTE_IRQHandler as well and they use the same port.

It'd be really nice to use the nrfx library so that others can use more advanced features in the nrfx_gpiote stack.  In my use case, I'm performing pulse counting via PPI, Timer, and GPIOTE.  I am probably going to have to hack the WInterrupts.h to expose a way for me to get an available channel in the handler and use the legacy driver specifically for GPIOTE.

Now... I'm not an nRF52840 expert, so maybe I'm wrong about this... but please correct me if I am.  Thank you for the consideration.

henrygab, 2020-05-07:
@czuvich -- I think you are correct, but there may be a workaround.

### GPIOTE interrupt handler already defined

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/b03d6e5d814b48bcbaac39e38cb5f40e2b8e729f/cores/nRF5/WInterrupts.c#L180

The above line defines the implementation for the GPIOTE interrupt handler.  As there can be only one interrupt handler, it prevents the use of other GPIOTE interrupt handlers.

### Other bugs you'll likely run into

With the current release (0.21.0), `attachInterrupt()` would blindly use GPIOTE channels, even if you had previously configured them for other purposes.  This was fixed in #485 nine days ago (seven days before your opened this issue), but it's not made it into a release yet.  This fix will **_at least_** prevent the BSP from overwriting your configuration.....

### Potential workaround?

This workaround adds a eight-cycle delay (AHB runs at 16MHz vs. the CPU 64Mhz) due to the inter-peripheral signaling required.  This is less than the SoftDevice adds, so that may be OK for your use.

The nRF52840 has a peripheral called "EGU", for event generator unit.  (_EGU1 is reserved for use by SoftDevice._)  As an example, it appears that the following could work to get you another interrupt handler:

* Define an interrupt handler for EGU2 (SWI2_EGU2_IRQHandler) in your code
* Configure a PPI channel:  GPIOTE event --> EGU2 TASKS_TRIGGER[N] registers

This *should* give you an ISR that you own, triggered by the GPIOTE event (just a few cycles later).

### Do you need an interrupt handler?

I'm guessing that you've configured the GPIOTE to trigger on either rising or falling edge, and connected it to a PPI channel, which itself triggers the COUNT task on a TIMER.

Are you sure this even requires a software interrupt?

It looks like it would all run without any CPU intervention.  You could even setup a second TIMER (configured as a timer) to automatically start/stop the measurements with high timing accuracy ... and only the initial start would require CPU involvement?

### Caveats

I am also investigating the use of PPI + TIMERS + GPIOTE + other peripherals.  However, I've only begun looking at them, so some things are odd.

One interesting thing: PPI will get signaled regardless of whether you have interrupts enabled on the peripheral.  Look at the block diagram of the generic peripheral:
[image]

As you can see, the event output from the peripheral core gets fed to PPI, whether the interrupt is enabled or not.  I didn't realize this for quite a while, so I thought I'd point that out, in case it can save you time also.

, if you need interrupts from the GPIOTE you're using, then I think your only option is to wire it up to a software interrupt (I don't think the current release of the BSP includes this fix yet, At the moment, there is no safe way to use GPIOTE.  The good news is that I also have wanted to use GPIOTE, and so have started the work needed to at least prevent `attachInterrupt()` from using a GPIOTE channel that's already in use.

, and it's not defined with the weak attribute.  This is used by `attachInterrupt()` (and maybe others).  I don't think the nrfx library is fully supported using this BSP ... for better or worse, it seems to have taken only some parts of nrfx.

czuvich, 2020-05-07:
@henrygab First of all thank you so much for the very detailed explanation!  Here are my responses:

> With the current release (0.21.0), attachInterrupt() would blindly use GPIOTE channels

Correct.  I basically hacked the WInterrupts class by adding a method which retrieves the assigned channel for the specified pin.  It just loops through the array just like `attachInterrupt()`.

> Are you sure this even requires a software interrupt?

I don't need a software interrupt; however, the code that generates the error is the following:
`nrfx_gpiote_in_event_addr_get(IN_PIN)` in combination with any calls to `attachInterrupt()`
Here's a link that shows basically what I'm doing, but using the legacy driver instead of nrfx.

[https://jimmywongiot.com/2019/10/03/how-to-count-gpio-pin-raising-falling-with-timer-counter-through-ppi/](url)

You are correct in that you use PPI and a TIMER, but you still need to assign the GPIOTE event to PPI which is where the issue comes in.  For instance...

```
nrfx_err_t err_code;
  if (!nrfx_gpiote_is_init()) {
    err_code = nrfx_gpiote_init(NRFX_GPIOTE_DEFAULT_CONFIG_IRQ_PRIORITY);
  }

  nrfx_gpiote_in_config_t in_config = NRFX_GPIOTE_CONFIG_IN_SENSE_LOTOHI(false);
  in_config.pull = NRF_GPIO_PIN_PULLUP;

  err_code = nrfx_gpiote_in_init(GC_WATER_FLOW_SIGNAL_PIN, &in_config, NULL);
  nrfx_gpiote_in_event_enable(GC_WATER_FLOW_SIGNAL_PIN, false);

  // ... TIMER initialization stuff omitted

 err_code = nrfx_ppi_channel_alloc(&ppi_channel_1);
  err_code = nrfx_ppi_channel_alloc(&ppi_channel_2);
  uint32_t gpiote_evt_addr_1 = nrfx_gpiote_in_event_addr_get(GC_WATER_FLOW_SIGNAL_PIN);

// PPI omitted
err_code = nrfx_ppi_channel_assign(_ppiChannel_1, gpiote_evt_addr_1, timer_count_count_task_addr);

 // all of this works.. UNTIL
attachInterrupt(BTN_PIN, handler, FALLING);
// comment the above attachInterrupt.. no error
```

I've tried various combinations of omitting various gpiote code, but the `nrfx_gpiote_in_event_addr_get` is most definitely the culprit.  I'm wondering if it's actually an issue with the NRF SDK itself.  

In the end, what I do is the following which seems to work, but I'm still not entirely sure if it's generating a software interrupt (I don't think it is as long as I'm routing the event directly to PPI).

```
pinMode(GC_WATER_FLOW_SIGNAL_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(GC_WATER_FLOW_SIGNAL_PIN), NULL, FALLING);
// instead of GPIOTE initialization, I use the attachInterrupt to add to the handler in the WInterrupts class

// .. TIMER stuff omitted
 err_code = nrfx_ppi_channel_alloc(&_ppiChannel_1);
  err_code = nrfx_ppi_channel_alloc(&_ppiChannel_2);

// requires the nRF pin, not the Arduino mapped pin, nrf_getChannelPort is my "hacked" in method which iterates over the array in WInterrupts
  int8_t channel = nrf_getChannelPort(GC_WATER_FLOW_SIGNAL_NRF_PIN); 

  nrf_gpiote_event_t gpiote_evt = nrf_gpiote_in_event_get((uint8_t)channel);
  uint32_t gpiote_evt_addr_1 = nrf_gpiote_event_address_get(NRF_GPIOTE, gpiote_evt);

err_code = nrfx_ppi_channel_assign(_ppiChannel_1, gpiote_evt_addr_1, timer_count_count_task_addr);
// PPI omitted
```

So, now I can safely use attachInterrupt() along with gpiote handling for counting pulses.  It still feels like a hack to me, but I can't think of a cleaner way around it.

> The nRF52840 has a peripheral called "EGU", for event generator unit. (EGU1 is reserved for use by SoftDevice.) As an example, it appears that the following could work to get you another interrupt handler:

I may have to look into that, but for now I think the hack works.  I'm definitely open to your suggestions and feedback!  Thanks.

czuvich, 2020-05-07:
I also want to add, it would be really nice to add more nrfx implementation to the BSP file.  I had to add the ppi, timer, and gpiote implementations.  I also had to add the necessary defines in the nrfx config file.  Overall, not hard thanks to how the project is structured, but I'm not entirely sure why nrfx implementations are omitted.

eugeneanikin, 2020-09-07:
I'd like to second pusselment about missing nrfx implementations. I had to re-add nrfx_timer.c and nrfx_pwm.c needed for my project. Any reasons why these files cannot be re-added from the SDK?

hathach, 2020-09-07:
adding nrfx files are OK, as long as it doesn't break anything and have no modification i.e stock nrfx files, this is to make upgrading easier. If you could submit an PR, I can give it a try.

eugeneanikin, 2020-09-07:
PR? Sorry,  git newbie here. Just to clarify, I need to make a branch with changes, commit and then file a pull request. Right? I'll be happy to do so. Could you point me to the exact version and configuration of the SDK used originally, so I get the right files.
Eugene

hathach, 2020-09-08:
- github has a details docs for submitting pr https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request
- current nrfx version is 2.1.0 https://github.com/NordicSemiconductor/nrfx/releases/tag/v2.1.0