adafruit/Adafruit_nRF52_Arduino #500: Adafruit_LittleFS.cpp:159:45: error: variable-sized object 'parent' may not be initialized

kylongmu, 2020-05-07:
**Describe the bug**
variable-sized object 'parent' may not be initialized

**Set up (please complete the following information)**
  - Adafruit_nRF52_Arduino version 0.20.1 - 2020.04.23
  - OS (windows10)
  - Arduino IDE 1.8.12

**To Reproduce**
Steps to reproduce the behavior:
1. Set Board as 'Adafruit Feather nRF52832'
2. Open 'beacon' example
3. Click on 'veryfy‘
4. See error

C:\Users\THINKPAD\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino\libraries\Adafruit_LittleFS\src\Adafruit_LittleFS.cpp: In member function 'bool Adafruit_LittleFS::mkdir(const char*)':

C:\Users\THINKPAD\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino\libraries\Adafruit_LittleFS\src\Adafruit_LittleFS.cpp:159:45: error: variable-sized object 'parent' may not be initialized

     char parent[slash - filepath + 1] = { 0 };

pyro9, 2020-07-17:
Could not verify using Arduino IDE 1.8.13 and Adafruit_nRF52_Arduino 0.20.5, it might be worth updating and trying it again.

hathach, 2020-09-09:
closed since there is no response from OP

closed 2020-09-09 by hathach

quentinmit, 2021-09-26:
I just experienced the same issue after updating to the latest `master`. Here's the more complete compilation output:

```
/Users/quentin/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=64000000 -DARDUINO=10815 -DARDUINO_NRF52840_FEATHER_SENSE -DARDUINO_ARCH_ADAFRUIT_NRF52_ARDUINO "-DARDUINO_BSP_VERSION=\"1.1.0\"" -DNRF52840_XXAA -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8087 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather nRF52840 Sense\"" -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DDX_CC_TEE -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -DCFG_LOGGER=1 -DCFG_SYSVIEW=0 -I/Users/quentin/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0/CMSIS/Core/Include/ -I/Users/quentin/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0/CMSIS/DSP/Include/ -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/nrfx -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/nrfx/hal -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/nrfx/mdk -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/nrfx/soc -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/nrfx/drivers/include -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/nrfx/drivers/src -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf52 -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/freertos/Source/include -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/freertos/config -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/freertos/portable/GCC/nrf52 -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/freertos/portable/CMSIS/nrf52 -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/sysview/SEGGER -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/sysview/Config -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_TinyUSB_Arduino/src/arduino -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5 -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/variants/feather_nrf52840_sense -I/Users/quentin/Documents/Arduino/libraries/Adafruit_NeoPixel -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Bluefruit52Lib/src -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_nRFCrypto/src -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_TinyUSB_Arduino/src -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/BLEAdafruitService/src -I/Users/quentin/Documents/Arduino/libraries/Adafruit_Unified_Sensor -I/Users/quentin/Documents/Arduino/libraries/Adafruit_GFX_Library -I/Users/quentin/Documents/Arduino/libraries/Adafruit_SSD1306 -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/SPI -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Wire -I/Users/quentin/Documents/Arduino/libraries/Adafruit_INA219 -I/Users/quentin/Documents/Arduino/libraries/Adafruit_BusIO -I/Users/quentin/Documents/Arduino/libraries/Adafruit_APDS9960_Library -I/Users/quentin/Documents/Arduino/libraries/Adafruit_BMP280_Library -I/Users/quentin/Documents/Arduino/libraries/Adafruit_LIS3MDL -I/Users/quentin/Documents/Arduino/libraries/Adafruit_LSM6DS -I/Users/quentin/Documents/Arduino/libraries/Adafruit_SHT31_Library -I/Users/quentin/Documents/Arduino/libraries/Adafruit_SPIFlash/src -I/Users/quentin/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src -I/Users/quentin/Documents/Arduino/libraries/Adafruit_AHRS/src -I/Users/quentin/Documents/Arduino/libraries/Adafruit_Sensor_Calibration -I/Users/quentin/Documents/Arduino/libraries/ArduinoJson/src -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_LittleFS/src -I/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/InternalFileSytem/src /Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS.cpp -o /var/folders/kn/0gfzt13x37j63313z_5szy3h0000gn/T/arduino_build_531816/libraries/Adafruit_LittleFS/Adafruit_LittleFS.cpp.o
/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS.cpp: In member function 'bool Adafruit_LittleFS::mkdir(const char*)':
/Users/quentin/Documents/Arduino/hardware/adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS.cpp:161:45: error: variable-sized object 'parent' may not be initialized
     char parent[slash - filepath + 1] = { 0 };
                                             ^
```

This is with Arduino 1.8.15. Currently about to update to 1.8.16 to see if that fixes it.

hathach, 2021-09-27:
Your gcc is too old, try to reinstall bsp, if you are clone with git. Make sure you do follow the guide in the readme

quentinmit, 2021-09-27:
@hathach Actually it seems that the Arduino IDE keeps bouncing between the 2014 GCC in the `arduino` package and the 2019 GCC in the `adafruit` package. It seems to flip back and forth randomly every time I try to compile the sketch. I do have the new GCC installed as well.