adafruit/Adafruit_nRF52_Arduino #502: pgm_read_ptr(addr) macr obuf

orrmany, 2020-05-10:
**Describe the bug**
pgm_read_ptr(addr) is faulty, causing PAul Stoffgren's Time library (among others) to not compile. 

**Set up (please complete the following information)**
  - Your Board and BSP version: nRF52840 Feather Express w
  - Your OS (mac/windows/linux) and its version: Windows 10
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2): not applicable since it is compile time error.

**To Reproduce**
Include <TimeLib.h> from ttps://github.com/PaulStoffregen/Time and try to compile it

**Expected behavior**
It shall compile 

**Screenshots**
n/a

**Additional context**
The compilation fails with this error message:
In file included from C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/Arduino.h:17:0,
                 from C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp:12:
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp: In function 'char* monthStr(uint8_t)':
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:105:49: error: 'const void*' is not a pointer-to-object
type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp:72:29: note: in expansion of macro 'pgm_read_ptr'
     strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(monthNames_P[month])));
                             ^~~~~~~~~~~~
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp: In function 'char* dayStr(uint8_t)':
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:105:49: error: 'const void*' is not a pointer-to-object
type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp:86:28: note: in expansion of macro 'pgm_read_ptr'
    strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(dayNames_P[day])));
                            ^~~~~~~~~~~~
*** [.pio\build\adafruit_feather_nrf52840_s340\libbe2\Time_ID44\DateStrings.cpp.o] Error 1
=============================================================================== [FAILED] 
`

orrmany, 2020-05-10:
Proposed fix is in PullRequest #503

hathach, 2020-07-13:
fixed by #503

closed 2020-07-13 by hathach