adafruit/Adafruit_nRF52_Arduino #506: Adafruit nRF52 in PlatformIO library registry

TorbenSchreiter, 2020-05-13:
**Is your feature request related to a problem? Please describe.**
Building a multi-environment project in PlatformIO (nRF52 + ESP32). Out of the Adafruit_nRF52_Arduino repository we are using the Bluefruit52Lib.

In PlatformIO I am struggling to integrate the lib without sideeffects. An automatic Github pull seems not to work due to the lib being in a submodule of the Adafruit_nRF52_Arduino project. Pulling it in platformio.ini like this:
`lib_deps = https://github.com/adafruit/Adafruit_nRF52_Arduino.git#master`

...gives strange errors:
```
PACKAGES:
 - framework-arduinoadafruitnrf52 1.1900.200401 (19.0)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing Adafruit_nRF52_Arduino
git version 2.26.0.windows.1
Cloning into 'C:\Users\[...]\.pio\libdeps\adafruit_feather_nrf52832\_tmp_installing-j_4lcok1-package'...
Submodule 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' (https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore.git) registered for path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'
Cloning into 'C:/Users/[...]/.pio/libdeps/adafruit_feather_nrf52832/_tmp_installing-j_4lcok1-package/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'...
fatal: '$GIT_DIR' too big
fatal: clone of 'https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore.git' into submodule path 'C:/Users/[...]/.pio/libdeps/adafruit_feather_nrf52832/_tmp_installing-j_4lcok1-package/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' failed
Failed to clone 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'. Retry scheduled
Cloning into 'C:/Users/[...]/.pio/libdeps/adafruit_feather_nrf52832/_tmp_installing-j_4lcok1-package/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'...
fatal: '$GIT_DIR' too big
fatal: clone of 'https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore.git' into submodule path 'C:/Users/[...]/.pio/libdeps/adafruit_feather_nrf52832/_tmp_installing-j_4lcok1-package/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' failed
Failed to clone 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' a second time, aborting
```

Directly addressing only the Bluefruit52Lib submodule in platformio.ini does not seem to work either.

**Describe the solution you'd like**
An Adafruit-maintained version of the Bluefruit52Lib library in the PlatformIO library registry would be great. There are many other official Adafruit libs in there already. Adding this one would most likely solve the above issues.

**Describe alternatives you've considered**
see above

**Additional context**
[none]

ladyada, 2020-05-13:
Hi this error seems to be in platformio - the $GIT_DIR variable is not one we set
please open an issue there, @ivankravets will be able to help you :)

TorbenSchreiter, 2020-05-13:
Main main point was, that I do not understand why the other Adafruit libs are in the registry but Bluefruit52Lib is not. Are you considering adding it?

ladyada, 2020-05-13:
they are not added by us at all, we publish to the Arduino library manager only. You could add it if you like?

TorbenSchreiter, 2020-05-13:
> they are not added by us at all, we publish to the Arduino library manager only. You could add it if you like?

This looks rather official to me:

[image]

Total of 225 libraries maintained by the author "Adafruit":
[image]

I am not interested personally to update the libraries myself in the future.

ladyada, 2020-05-13:
yeah those all come from the arduino library manager, we've never submitted libraries ourselves to platformio.

ivankravets, 2020-05-13:
@ladyada, PlatformIO does not use Arduino tools, services and does not depend on them. PlatformIO Library Registry has been working since 2014, where Arduino.cc added support only in 2016 or 2017, I don't remember.

There is also some attack by @mbanzi https://twitter.com/mbanzi/status/1257980326808338434

-------

Arduino IDE lib manager has support for 3k libraries. PlatformIO has support for over 8k. Developers use [library.json](https://docs.platformio.org/en/latest/librarymanager/config.html)  instead of library.properties and submit libraries directly to PlatformIO's Registry.

> we've never submitted libraries ourselves to platformio.

We have Github crawler which indexes public repositories and looks for embedded libraries (library.json ESP-IDF components, mbed.json, Zephyr modules, library.properties). This is a way how adafruit's libs were autoregistered in PlatformIO. If some libs should be removed, please contact us.

Thanks.

ladyada, 2020-05-13:
hi @ivankravets, ok good to know we will let people know that the libraries are auto-crawled & auto-registered. @TorbenSchreiter  we do not manually submit any libriares to the platformio manager

@ivankravets please see the @TorbenSchreiter  issue, something about GIT_DIR path being too long.

TorbenSchreiter, 2020-05-13:
Ok, @ladyada's hint and this thread (https://community.platformio.org/t/pio-compiling-error-for-nrf52832/9680/4) clarified the dependencies better. There is no need to add the `Bluefruit52Lib` at all, as it is already included in the PIO platform package `Nordic nRF52`.

It was a bit confusing in the first place that PIO listed the auto-crawled/-registered `Adafruit BluefruitLE nRF51` lib, but that's clear now. I have the correct lib in my includepath now.

However, I am still experiencing dozens of build issues, that are somewhat strange and only seem to relate to library code:

```
Processing adafruit_feather_nrf52832 (platform: nordicnrf52; board: adafruit_feather_nrf52832; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52832.html
PLATFORM: Nordic nRF52 4.2.0 > Adafruit Bluefruit nRF52832 Feather
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.1900.200401 (19.0)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <Tasker> 2.0
|-- <MLX90640_API>
|   |-- <Wire> 1.0
|-- <Adafruit Bluefruit nRF52 Libraries> 0.19.0
|   |-- <Adafruit Little File System Libraries> 0.11.0
|   |-- <Adafruit Internal File System on Bluefruit nRF52> 0.11.0
|   |   |-- <Adafruit Little File System Libraries> 0.11.0       
|-- <VL53L0X-2xI2C>
|   |-- <Wire> 1.0
|-- <spline>
Building in release mode
Compiling .pio\build\adafruit_feather_nrf52832\src\ble.cpp.o
Compiling .pio\build\adafruit_feather_nrf52832\src\main.cpp.o
src\ble.cpp:11:28: warning: integer constant is too large for its type
   mainService = BLEService(0x00000001000000fd8933990d6f411ff7);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/bluefruit.h:46:0,
                 from src\ble.h:11,
                 from src\ble.cpp:1:
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:44:5: error: 'ble_uuid_t' does not name a type; did you mean '__uid_t'?
     ble_uuid_t _uuid;
     ^~~~~~~~~~
     __uid_t
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:51:24: error: expected ')' before 'uuid'
     BLEUuid(ble_uuid_t uuid           ) { _uuid = uuid; _uuid128 = NULL; }
                        ^~~~
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:66:27: error: 'ble_uuid_t' does not name a type; did you mean '__uid_t'?
     bool operator==(const ble_uuid_t uuid) const;
                           ^~~~~~~~~~
                           __uid_t
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:67:27: error: 'ble_uuid_t' does not name a type; did you mean '__uid_t'?
     bool operator!=(const ble_uuid_t uuid) const;
                           ^~~~~~~~~~
                           __uid_t
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:72:24: error: declaration of 'operator=' as non-function
     BLEUuid& operator=(ble_uuid_t uuid);
                        ^~~~~~~~~~
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:72:22: error: expected ';' at end of member declaration
     BLEUuid& operator=(ble_uuid_t uuid);
                      ^
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:72:35: error: expected ')' before 'uuid'
     BLEUuid& operator=(ble_uuid_t uuid);
                                   ^~~~
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h: In constructor 'BLEUuid::BLEUuid()':
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/BLEUuid.h:48:43: error: '_uuid' was not declared in this scope
[... 599 lines cut ...]
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/bluefruit.h:221:5: error: 'ble_gap_sec_params_t' does not name a type
     ble_gap_sec_params_t _sec_param;
     ^~~~~~~~~~~~~~~~~~~~
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/bluefruit.h:234:24: error: 'ble_evt_t' has not been declared
     void (*_event_cb) (ble_evt_t*);
                        ^~~~~~~~~
C:\Users\[...]\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib\src/bluefruit.h:244:23: error: 'ble_evt_t' has not been declared
     void _ble_handler(ble_evt_t* evt);
                       ^~~~~~~~~
src\main.cpp: In function 'void setup()':
src\main.cpp:78:54: warning: embedded '\0' in format [-Wformat-contains-nul]
[... cut ...]
Environment                Status    Duration
-------------------------  --------  ------------
lolin_d32                  IGNORED
adafruit_feather_nrf52832  FAILED    00:00:11.746
============================================================================================= 1 failed, 0 succeeded in 00:00:11.746 =============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

ladyada, 2020-05-13:
@TorbenSchreiter ok good you found out the source of the issue, please work with @ivankravets to determine how to change the compiler!

ivankravets, 2020-05-13:
@TorbenSchreiter please forward your issue and build log to https://community.platformio.org/c/development-platforms/17

TorbenSchreiter, 2020-05-13:
Thanks guys. It was a plain file naming conflict. Bluefruit52Lib is referencing Nordic's `ble.h` (https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/nordic/softdevice/s132_nrf52_6.1.1_API/include/ble.h).

The project had another `ble.h`. I changed the filename and it compiles.

closed 2020-05-13 by TorbenSchreiter