adafruit/Adafruit_nRF52_Arduino #507: Parings/Bondings are lost during flash with adafruit-nrfutil

dzervas, 2020-05-15:
**Describe the bug**
Every time that I flash the Bluefruit Feather, I have to remove the device from my computer and re-pair. I speculate that this is caused due to an EEPROM erase? Maybe I need to pass a flag to nrfutil to avoid erasing the flash.

**Set up (please complete the following information)**
  - Bluefruit Feather nrf52832
  - Tested on Linux, Mac & iOS

**To Reproduce**
1. Pair with computer
2. Reflash the board
3. OS can't reconnect and the device need to be re-paired

**Expected behavior**
The board should keep the bondings across flashes

hathach, 2020-05-16:
Can you tell me which sketch that you use, the update shouldn’t erase the location that stores the bonding. I also need the bsp version and serial logs with debug level = 2 where it failed to reconnect as well.

dzervas, 2020-05-18:
```
1587186770.256204: 1587186770.256216: wlan0_1: STA 64:70:33:**:**:** IEEE 802.11: deauthenticated due to i
BSP Library : 0.19.0
Bootloader  : s132 6.1.1
Serial No   : 325D0652143B96CA
1587187101.315511: 1587187101.315520: wlan0_1: STA 64:70:33:**:**:** IEEE 802.11: associated (aid 4)
[CFG   ] SoftDevice's RAM requires: 0x20002930
lfs error:494: Corrupted dir pair at 3 2
lfs error:494: Corrupted dir pair at 3 2
bool Adafruit_LittleFS::mkdir(const char*): 165: verify failed, error = LFS_ERR_CORRUPT
lfs error:494: Corrupted dir pair at 3 2
lfs error:494: Corrupted dir pair at 3 2
bool Adafruit_LittleFS::mkdir(const char*): 165: verify failed, error = LFS_ERR_CORRUPT
--------- SoftDevice Config ---------
Max UUID128     : 10
ATTR Table Size : 3072
Service Changed : 1
Peripheral Connect Setting
  - Max MTU         : 23
  - Event Length    : 3
  - HVN Queue Size  : 1
  - WrCmd Queue Size: 1

--------- BLE Settings ---------
Name            : Plikter
Max Connections : Peripheral = 1, Central = 0
Address         : F1:BF:D6:**:**:** (Static)
TX Power        : -20 dBm
Conn Intervals  : min = 11.25 ms, max = 15.00 ms
Conn Timeout    : 2000 ms
Peripheral Paired Devices:
lfs error:494: Corrupted dir pair at 3 2
bool Adafruit_LittleFS_Namespace::File::_open(const char*, uint8_t): 136: verify failed, error = LFS_ERR_CORRUPT

[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 15.00 ms, Latency = 0, Supervisor Timeout = 2000 ms
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 0
[GAP   ] Data Length Request is (tx, rx) octets = (123, 123), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 0
[GAP   ] Data Length is (tx, rx) octets = (27, 27), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_SEC_INFO_REQUEST : Conn Handle = 0
lfs error:494: Corrupted dir pair at 3 2
bool Adafruit_LittleFS_Namespace::File::_open(const char*, uint8_t): 136: verify failed, error = LFS_ERR_CORRUPT
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[GAP   ] Disconnect Reason: REMOTE_USER_TERMINATED_CONNECTION
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 15.00 ms, Latency = 0, Supervisor Timeout = 2000 ms
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 0
[GAP   ] Data Length Request is (tx, rx) octets = (123, 123), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 0
[GAP   ] Data Length is (tx, rx) octets = (27, 27), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 0
[GAP   ] ATT MTU is changed to 23
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0
```

dzervas, 2020-05-19:
After doing:
```
        InternalFS.begin();
        InternalFS.format();
        Bluefruit.clearBonds();
        Bluefruit.Central.clearBonds();
```

I now get:

```
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 15.00 ms, Latency = 0, Supervisor Timeout = 2000 ms
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 0
[GAP   ] Data Length Request is (tx, rx) octets = (123, 123), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 0
[GAP   ] Data Length is (tx, rx) octets = (27, 27), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_SEC_INFO_REQUEST : Conn Handle = 0
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[GAP   ] Disconnect Reason: REMOTE_USER_TERMINATED_CONNECTION
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 15.00 ms, Latency = 0, Supervisor Timeout = 2000 ms
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 0
[GAP   ] Data Length Request is (tx, rx) octets = (123, 123), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 0
[GAP   ] Data Length is (tx, rx) octets = (27, 27), (tx, rx) time = (1096, 1096) us
[BLE   ] BLE_GAP_EVT_SEC_INFO_REQUEST : Conn Handle = 0
...
```

dzervas, 2020-05-19:
OK, seems that maybe I forgot to re-pair after `clearBonds`? Works for now but I smell problems...

dzervas, 2020-05-19:
(closed by mistake)

closed 2020-05-19 by dzervas