adafruit/Adafruit_nRF52_Arduino #512: Use Capslock, Ctrl, Alt on BLE HID-Keyboard

deepkap, 2020-05-25:
How to send key press event of Capslock, Ctrl, Alt keys via BLE HID on nRF52840 Express?

jpconstantineau, 2020-05-25:
These are all different "keycodes" that can be sent. To see a simple
example, have a look at the "hid_keyscan"  example.  I used this example as
the starting point for the BlueMicro_BLE keyboard firmware.

If you want a simple keyboard, you can stick with the example.  If you want
a complex keyboard with lots of functionality like layers, macros and lots
of other features that one would expect from a fully-featured programmable
keyboard, have a look at my BlueMicro_BLE repo.
It builds on top of the Adafruit_nRF52_Arduino to build a keyboard
firmware. Thanks Adafruit and all contributors for supporting the BSP -
your work makes mine much simpler as I can focus on my features.

Also, questions such as this should go to the Adafruit forums.

hathach, 2020-05-25:
yes, for more questions and how-to please ask on the support forum

closed 2020-05-25 by hathach