adafruit/Adafruit_nRF52_Arduino #514: Discovery failed on Manufacture Characteristic in Device Infomation Service.

arms22, 2020-05-27:
**Describe the bug**
Discovery failed on Manufacture Characteristic in Device Infomation Service.

**Set up (please complete the following information)**
  - Raytac MDBT50Q-RX Dongle
  - Windows 10
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  
```text
[GAP   ] Conn Interval= 20.00 ms, Latency = 0, Supervisor Timeout = 2000 ms
[BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 0
[DISC  ] [SVC] Handle start = 1
[BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 0
[GAP   ] ATT MTU is changed to 131
[BLE   ] BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP : Conn Handle = 0
[DISC  ] [SVC] Service Count: 1
[DISC  ] [SVC] Found 0x180A, Handle start = 14, end = 24    <-- Device Infomation Service
-----------------
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A24, handle = 16 
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A25, handle = 18
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A26, handle = 20
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Handle start = 8961, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 0
[DISC  ] [CHR] Gatt Status = 0x0101    <-- Invalid Attribute Handle
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A28, handle = 22
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Handle start = 8961, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 0
[DISC  ] [CHR] Gatt Status = 0x0101    <-- Invalid Attribute Handle
```

**Additional context**
When the Characteristic Count is more than 5, I think the cause is that the access exceeds the memory area allocated by rtos_malloc.

This is where the memory for four characteristic  is allocated.
```
uint8_t BLEDiscovery::discoverCharacteristic(uint16_t conn_handle, BLEClientCharacteristic* chr[], uint8_t count)
{
  // We could found more characteristic than we looking for. Buffer must be large enough
  enum { MAX_DISC_CHARS = 4 };

  uint16_t bufsize = sizeof(ble_gattc_evt_char_disc_rsp_t) + (MAX_DISC_CHARS-1)*sizeof(ble_gattc_char_t);
  ble_gattc_evt_char_disc_rsp_t* disc_chr = (ble_gattc_evt_char_disc_rsp_t*) rtos_malloc( bufsize );
```

When the value of d is 5, it reads an invalid value.
```
    // Look for matched uuid in the discovered list
    for(uint8_t d=0 ; d<disc_chr->count; d++)
    {
      for (uint8_t i=0; i<count; i++)
      {
        // Skip if output chr is already discovered, happens with multiple instances of same UUIDs
        if ( chr[i]->discovered() ) continue;

        if ( chr[i]->uuid == disc_chr->chars[d].uuid )
```

pyro9, 2020-07-13:
It looks like at LEAST the buffsize needs to be set based on the handle range (set when the parent service is discovered), otherwise no service with more than 4 characteristics will ever have all of its characteristics discovered.

In truth, the whole discovery process probably needs a bit of an overhaul. In actual use, it seems that characteristics might be discovered one at a time, causing BLEDiscovery::discoverCharacteristic to run for each characteristic getting the same answer back from sd_ble_gattc_characteristics_discover each time. That should probably be cached until the next service is discovered.

pyro9, 2020-07-13:
What peripheral device are you discovering? It looks like the device(s) BLEDiscovery was tested against just happen to answer discovery requests in chunks of 3.

pyro9, 2020-07-13:
@arms22 can you try the following patch? 
```
--- a/libraries/Bluefruit52Lib/src/BLEDiscovery.cpp
+++ b/libraries/Bluefruit52Lib/src/BLEDiscovery.cpp
@@ -109,7 +109,8 @@ uint8_t BLEDiscovery::discoverCharacteristic(uint16_t conn_handle, BLEClientChar
   // We could found more characteristic than we looking for. Buffer must be large enough
   enum { MAX_DISC_CHARS = 4 };
 
-  uint16_t bufsize = sizeof(ble_gattc_evt_char_disc_rsp_t) + (MAX_DISC_CHARS-1)*sizeof(ble_gattc_char_t);
+  // -1 because the first ble_gattc_char_t is built in to ble_gattc_evt_char_disc_rsp_t
+  uint16_t bufsize = sizeof(ble_gattc_evt_char_disc_rsp_t) + (MAX_DISC_CHARS-1)*sizeof(ble_gattc_char_t); 
   ble_gattc_evt_char_disc_rsp_t* disc_chr = (ble_gattc_evt_char_disc_rsp_t*) rtos_malloc( bufsize );
 
   uint8_t found = 0;
@@ -129,6 +130,9 @@ uint8_t BLEDiscovery::discoverCharacteristic(uint16_t conn_handle, BLEClientChar
     // timeout or has no data (due to GATT Error)
     if ( bytecount <= 0 ) break;
 
+    if ( disc_chr->count > MAX_DISC_CHARS ) // if we truncated the response, adjust the count to match
+      disc_chr->count = MAX_DISC_CHARS;
+
     // Look for matched uuid in the discovered list
     for(uint8_t d=0 ; d<disc_chr->count; d++)
     {
```

hathach, 2020-07-14:
thanks for filing the issue, can you provide a bit more information
- Your BSP version
- What device you are discovering
- The real attribute table of the target device. You can get this by using nRFConnect desktop or Mobile app from Nordic.

Since I don't have you device to reproduce, I will mostly help with analysis the data.

pyro9, 2020-07-14:
I don't have the device either, but I tested by reducing MAX_DISC_CHARS to 2 and using the bleuart and bleuart_central examples. I used the BSP pulled yesterday (07-13-2020).

The fundamental issue is that disc_chr->count can be larger than MAX_DISC_CHARS and in that case, the inner loop of BLEDiscovery::discoverCharacteristic will walk past the end of the buffer and discovery will fail for some characteristics.

This probably wasn't seen in testing since peripherals using this BSP will never return more than 3 characteristics at a time, but other devices apparently do (based on @arms22's debug log).

See PR #535

arms22, 2020-07-15:
Hi pyro9 and hathach.

> Your BSP version
0.20.1

> What device you are discovering
Custom boards we are developing.

> The real attribute table of the target device. You can get this by using nRFConnect desktop or Mobile app from Nordic.
Please see the attachment.

To pyro9

After applying the patch, discovery is now successful.
```
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 20.00 ms, Latency = 0, Supervisor Timeout = 2000 ms
[BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 0
[DISC  ] [SVC] Handle start = 1
[BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 0
[GAP   ] ATT MTU is changed to 131
[BLE   ] BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP : Conn Handle = 0
[DISC  ] [SVC] Service Count: 1
[DISC  ] [SVC] Found 0x180A, Handle start = 14, end = 24
-----------------
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A24, handle = 16
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A25, handle = 18
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A26, handle = 20
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Handle start = 23, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 1
[DISC  ] [CHR] Handle start = 25, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 0
[DISC  ] [CHR] Gatt Status = 0x0101
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Found 0x2A28, handle = 22
-----------------
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
[DISC  ] [CHR] Handle start = 14, end = 24
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[DISC  ] [CHR] Characteristic Count: 5
[DISC  ] [CHR] Handle start = 23, end = 24
```

[image]
[image]
[image]
[image]
[image]

hathach, 2020-07-16:
thanks for confirmation, @pyro9 did an excellent jobs to fix the issue even before all the information is obtained. Bravoos on that :+1: :+1:

hathach, 2020-07-17:
fixed by #535

closed 2020-07-17 by hathach