adafruit/Adafruit_nRF52_Arduino #521: BLE IMU Data capture

shandles, 2020-07-02:
Love this little board, I am trying to capture IMU data while moving the board in 3d space and using a USB cable is problematic.   It would be great to have a simple sketch to capture motion (using the user button D7) locally or via BLE transmission for ML algorithms.  I am currently working on this and will share an update once I have something user friendly.   If anyone out there has already done this, would love to see the code!

ladyada, 2020-07-02:
we have a guide here on fusing IMU data
https://learn.adafruit.com/how-to-fuse-motion-sensor-data-into-ahrs-orientation-euler-quaternions
it sounds like you want product/project help - not a bug in the BSP. in that case, please post to the support forums of the hardware you purchased. thanx!

closed 2020-07-02 by ladyada

shandles, 2020-07-02:
Thanks Ladyada for the quick reply!   I've been through the guide and it is very helpful..  my point is that data capture over BLE for the IMU is a little more complicated but immensely more useful / accessible than USB data capture...   I'll take a look at the support forums if I need any further help.    

Along these lines, I was unable to get the ST ML software (Unico) to work with this board - which would also make things easier....    

I'll happily post updates...