adafruit/Adafruit_nRF52_Arduino #526: Add function to allow changing I2C pins

thaanstad, 2020-07-07:
I sometimes want to change the I2C pins to other digital pins when interfacing with multiple devices with the same I2C address.  It is possible to instantiate a new TwoWire object and define the I2C pins in the constructor but many libraries utilize the pre-existing Wire object so it can be quite a hassle to modify these libraries and pass in a pointer to a new TwoWire object.  I have been modifying the TwoWire class with a setI2C pins function for some time now and that has made it easier to use these preexisting libraries while still being able to use other pins besides the default ones specified for the adafruit NRF582 board.

It seems like there is a lot of active development with this core lately so perhaps adding setI2CPins(int SDA, int SCL) or similar functionality would help others as well.  I have been adding this declaration in Wire.h and the following definition in Wire_nRF52.cpp.

void TwoWire::setI2CPins(int SDA, int SCL)
{
	this->_uc_pinSDA = g_ADigitalPinMap[SDA];
	this->_uc_pinSCL = g_ADigitalPinMap[SCL];
}

I think you may have to call Wire.begin() before using the other I2C device on different IO pins so perhaps there is a better alternative that I am not aware of?  Also the setI2CPins function I have been using could be expanded upon to configure the pins correctly without calling Wire.begin() again.

hathach, 2020-07-08:
That is a good reason to add the API, much similar to the uart one here 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/Uart.h#L37

Let's simply name it `setPins()`  if you don't mind submitting a PR for it, I am happy to review.

jpconstantineau, 2020-07-08:
I would love to see such functionality.  

When I implemented an OLED screen on my custom board, I had to do two things to map I2C to other gpios:
1 - In the top of the file:
TwoWire  Wire2(NRF_TWIM0, NRF_TWIS0, SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQn,15, 17 );

2 - in Setup:
Wire = Wire2;

It worked in my situation but I always felt it was a very clunky way to get it done.
Doing it the way proposed is a whole lot more elegant than what I did.  
Looking forward to see this as part of the BSP.

thaanstad, 2020-07-08:
I will update the code with the setPins() and submit a PR then.  I can try testing if calling Wire.begin() is required after switching the pins as well.

hathach, 2020-07-09:
as the general rule, the setPins() must be called before the begin().

hathach, 2020-08-20:
closed by #536

closed 2020-08-20 by hathach