adafruit/Adafruit_nRF52_Arduino #529: Pull up to 0.20.1 on PlatformIO

huddy987, 2020-07-12:
Hello,

I noticed when installing the BSP through PlatformIO (board = nrf52840_dk_adafruit), it downloads release 0.19. Any chance this could get pulled up to 0.20.1? I could always manually install the repo, but it is nice to get the updates through PlatformIO.

Thanks!

huddy987, 2020-07-12:
Just an update on this, to use the repo instead of whatever release PlatformIO decides, just copy paste this repo into the platformio packages/framework-arduinoadafruitnrf52 folder. Have to leave the package.json folder in there to trick PlatformIO into thinking its an official package.

hathach, 2020-07-13:
please post this to PlatformIO team, they should be able to update their cralwer/indexer

closed 2020-07-13 by hathach