adafruit/Adafruit_nRF52_Arduino #537: delay() wakes up prematurely and busy-waits

Nenik, 2020-07-15:
Slightly related to #165 

Using nRF52 package 0.20.1, measuring the power consumption on modified pca100059 (externally powered by LiIon to VddH pin, measuring the battery power, DCDC  on, but not relevant here)

While optimizing power consumption, I have noticed peculiar behavior of delay().
Consider trivial sketch of:
```
#define LED_AA 6

void setup() {
    // Enable the LED, turn it on
    pinMode(LED_AA, OUTPUT);
}

void loop() {
    for (;;) {
        delay(500);
        digitalToggle(LED_AA);
    }
}

```
I am getting great low power consumption during the sleep, but the CPU wakes up (observable by rise of supply current from uA to mA) about 1ms before the LED toggles (which obviously adds the LED current, but not relevant here either, just a diagnostic tool).

I have tried to trace the implementation deep into the FreeRTOS, and what I have found so far is that inside the idle task (function portTASK_FUNCTION), the sleep time is properly resolved and portSUPPRESS_TICKS_AND_SLEEP(time) is called. This causes proper CPU sleep and this function returns just when the CPU is woken up again (observed supply current starts to rise).
But then, the idle task will keep spinning for about 1ms (consuming about 5mA w/o DC/DC) before finally unblocking loop task, thus returning from the delay function.

This causes increased power consumption, since the CPU is awake 1ms more than would be necessary per delay call (i.e. in my example, 2ms per s, i.e. 10uA of average consumption).

I originally thought there is some off-by-one error and tried to extend the internal wait interval by 1ms, but that didn't affect the behavior (perhaps besides changing the rate).

Now I suspect that the issue is in the scheduler (will need to debug more to verify): perhaps after waking up, the loop thread is still not marked runnable and it needs a timer tick to actually schedule it. I am not much familiar with the FreeRTOS internals though.

hathach, 2020-07-15:
It is possible that FreeRTOS kernel want to wake up 1 tick earlier for doing overhead work (update tick, task switching etc ..) You probably want to go through their docs of the tickless mode.

PS: Since you are labeling this as bug, you must fill out all the questions in the template. Please update your 1st post accordingly.

Nenik, 2020-07-15:
> It is possible that FreeRTOS kernel want to wake up 1 tick earlier for doing overhead work (update tick, task switching etc ..) 
No, that's not the case. It will just spend that almost 1ms - till the next timer interrupt - looping inside the idle task doing nothing.

Looking deeper into the scheduler implementation:
Upon waking up from the sleep, a "clock update" happens - vTaskStepTick() gets called in the context of the idle task (since it was the idle task that forced the sleep).

What I see (for my `delay(500);` case) in the context of `vTaskStepTick()` function is:
xTicksToJump = 512 - right, 500ms with 1024Hz clock tick
xTickCount = 10419235 - current time
xNextTaskUnblockTime = 10419747 - that matches 10419235+512 - i.e. my task should wake up.

But there is nothing else besides time update in that function, no scheduling is performed.

I will update the template, though the issue currently looks more like FreeRTOS bug to me at this point.

hathach, 2020-07-15:
@Nenik maybe you could try to post this to Nordic devzone, we mostly use their freeRTOS tickless port example for this repo. Nordic should be able to provide an more insight information. Before posting there, it is best to try reproduce the issue with their sdk. Since they may not be familiar with this repo.

Nenik, 2020-07-16:
Done: https://devzone.nordicsemi.com/f/nordic-q-a/63828/vtaskdelay-on-nrf52-freertos-port-wastes-cpu-power

Nenik, 2020-07-16:
I have also found a workaround that seems to be working (described in the Nordic post), but I'll need more testing to validate it.

Nenik, 2020-08-09:
Nordic has quickly accepted the issue and proposed an alternative solution (on the same tune, just keeping the modification in the port code, as opposed to FreeRTOS core), see:
https://devzone.nordicsemi.com/f/nordic-q-a/63828/vtaskdelay-on-nrf52-freertos-port-wastes-cpu-power
Since it would likely take quite long before Nordic posts a new SDK with the fix, can we integrate the solution in advance for nRF52/BlueFruit core?

hathach, 2020-08-10:
@Nenik sure, would you mind doing an PR for this, It is much easier to review the changes with PR.

hathach, 2020-08-20:
fixed by pr #555